Denvi/Candle #391: Z axis wrong direction

TIPLDN, 2020-06-06:
Recently got my 3018pro & Set up my machine and tried to set up the correct settings with $3=4 but still when I click up it goes down and when I click down it goes up. How can I resolve this please?

ondras12345, 2020-06-07:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask

Try $3=0

TIPLDN, 2020-06-07:
Thank you that worked.

So I have started milling and it’s doing it in the opposite direction to what it should be, why would this be doing this? So say I have a square on a 3mm acrylic, the square should be on the right side but it’s milking it on the left? 

Also the first depth seems to be going deeper than it should?

Thanks

ondras12345, 2020-06-07:
Make sure your axes are moving in the correct directions. Refer to http://linuxcnc.org/docs/html/user/user-concepts.html#_machine_configurations . Make sure you understand the difference between tool directions and table directions. Use $3 to invert individual axes if required.

> the square should be on the right side but it’s milling it on the left?

The X axis direction is probably wrong. Try $3=1.

> Also the first depth seems to be going deeper than it should?

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#100-101-and-102--xyz-stepsmm
Please make sure to backup your current Grbl settings - the result of the `$$` command. In case your problems persist, post your settings and Grbl firmware version.

TIPLDN, 2020-06-07:
Ok so the machine is to the right side of me, I have put $3=0 which was making the axis’s all going in the right direction. When I put $3=1 the up, down, left, right are going opposite ways. 

I’ll looks at the tool and table directions now.

ondras12345, 2020-06-07:
There is a table in the [description of $2](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#2--step-port-invert-mask) that also applies to $3.

> I have put $3=0 which was making the axis’s all going in the right direction. When I put $3=1 the up, down, left, right are going opposite ways.

I don't think this is possible.

TIPLDN, 2020-06-07:
$3=0 goes in the right direction to the arrows when clicked. But when milling the rectangle I’m cutting should be in the right side but cutting in the left side. And I have put .1mm cutting depth but first layer is going deeper than that.

ondras12345, 2020-06-07:
> But when milling the rectangle I’m cutting should be in the right side but cutting in the left side.

Please post your Grbl settings and Gcode program.

TIPLDN, 2020-06-07:
My gbrl is not always opening why is this? I can’t get it to open now?

ondras12345, 2020-06-07:
Grbl is the firmware on your machine. The version is the fist message in the Console (e.g. 0.9j).

TIPLDN, 2020-06-07:
[image]
[image]

ondras12345, 2020-06-07:
Please post your settings and the gcode program as text. I can only see one of the images in your previous post.

TIPLDN, 2020-06-07:
The program is not opening now I had to close it because the commands wasn’t showing. It keeps doing this. I have to continuously restart my laptop and maybe after the 3rd restart it will run. Why is also happening?

ondras12345, 2020-06-07:
Try deleting Settings.ini (or rename it to something else).

TIPLDN, 2020-06-07:
$$ < $0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=400.000 (x, step/mm)
$101=400.000 (y, step/mm)
$102=400.000 (z, step/mm)
$110=1500.000 (x max rate, mm/min)
$111=1500.000 (y max rate, mm/min)
$112=1500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
ok

TIPLDN, 2020-06-07:
not sure how to copy the gcode command

TIPLDN, 2020-06-07:
also after changing the renaming the settings file the work space screens now black

TIPLDN, 2020-06-07:
No worries worked the screen out. How do I get the Gcode command in the console section

ondras12345, 2020-06-07:
Just post the gcode program file you are using to mill the square.

TIPLDN, 2020-06-07:
[.1mm MT Test.zip](https://github.com/Denvi/Candle/files/4742664/default.1mm.MT.Test.zip)

ondras12345, 2020-06-07:
For now, please set $3=0.
When you try to jog to the right, which direction does the tool move in? Repeat this for the other axes and post the results. When testing the Y axis, make sure you are watching the tool movement relative to the table and not the table relative to the frame of the machine.

TIPLDN, 2020-06-07:
When I jog to the right it goes right. When I jog down it goes down.

TIPLDN, 2020-06-07:
When I jog down the table moves down

ondras12345, 2020-06-07:
> When I jog down the table moves down

The table should move towards you when you press the Y+ button, because the tool is moving away from you relative to the table.

To invert the Y axis, use $3=2.

After you verify that the machine jogs in the correct directions, try running the gcode program again and see if it works.
Make sure to set the correct 0,0,0 point using the "Zero XY" and "Zero Z" buttons before running it.

TIPLDN, 2020-06-07:
Ok will have a go tomorrow. And what about the first initial layer being to deep to start?

ondras12345, 2020-06-08:
> And what about the first initial layer being to deep to start?

Check your steps/mm settings ($100-$102). Try moving the machine say 10 mm and measure the distance it actually moved.
Commands to do it:
```
G91
G0 X10
G90
```
This sets relative distance mode, moves the X axis 10mm to the right and sets absolute mode.
Repeat the same steps for all axes.

TIPLDN, 2020-06-08:
Ok all seems ok, but keeps disconnecting from the PC, why is this?

TIPLDN, 2020-06-08:
And is there a way to continue the work once connection is lost

ondras12345, 2020-06-08:
> Ok all seems ok, but keeps disconnecting from the PC, why is this?

https://github.com/Denvi/Candle/issues/342
Also, try running the USB cable further away from the rest of the wiring.

> And is there a way to continue the work once connection is lost

https://github.com/Denvi/Candle/issues/350

TIPLDN, 2020-06-08:
Im going to buy a new usb cable see if that helps and take it from there.

TIPLDN, 2020-06-09:
Hi there,

So I gone to test it now and managed to sorted the USB cable issue out. When I send the program the mill hit dragged through the acrylic, yesterday it lifted to a safe hight? Why is this?

TIPLDN, 2020-06-09:
[image]

ondras12345, 2020-06-09:
Line 6 of the gcode program (`G28 G91 Z0`) uses the [G28 command](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g28-g28.1). I'm not sure how this is implemented in Grbl, but it definitely can cause problems if you don't home your machine, because the origin (0,0,0) of machine coordinates (without homing) is just set when the machine starts, and thus can be different each time.

TIPLDN, 2020-06-20:
Hi there,

So I managed to sort most out, still having the gbhl loose connection 7/8mins into the milling. So I now have 2 other cables to try. 

But I noticed now that the measurements are off by a good few mill, how would I sort the settings to get the correct measurements on the system?

Thanks

cornnatron, 2020-06-27:
euhh you would have to calculate your steps of your stepper to get the right distance moved.
but what i would recommend more is first do some read up on the grbl wiki for the settings and also on linuxcnc.org to get a better 
understanding of the basics .