Denvi/Candle #393: no realtime correction of Z axis over long straight lines causing uneven depth

bubb1eb0b, 2020-06-12:
candle 1.1.7
GBRL 1.1h

hi
Candle does not perform realtime interpolation along straight lines
i make PCBs with long tracks (50 ohms microstrips)
with height map function activated, the Z position never changes during a single G01 move 
making the depth of carving uneven

as an exemple the Merlin firmware on my 3D printer performs realtime correction of Z position making Z offset perfectly even over all the bed, thanks to simple yet good bed levelling algorithms

at this moment the heightmap correction works for very little displacements or curves, but in case of an uneven surface with 0,1mm differences ovec differents points of the grid, the heightmap does not perform good

i think that is a really important feature for making RF PCBs, and for improving the precision in general

best regards

Denvi, 2020-06-12:
Hi, sample files & screenshots please.
Candle can divide long lines into smaller ones.

ondras12345, 2020-06-12:
What interpolation grid did you use? It seems to work ok for me.
Candle 1.1.7, Grbl 1.1h

[test1.txt](https://github.com/Denvi/Candle/files/4770229/test1.txt) - without heightmap
[test1t.txt](https://github.com/Denvi/Candle/files/4770230/test1t.txt) - with heightmap, 20x20 interpolation grid, 2x2 probe grid
[test1.map.txt](https://github.com/Denvi/Candle/files/4770249/test1.map.txt)