Denvi/Candle #398: Marlin protocol support

atlaste, 2020-06-29:
**Is your feature request related to a problem? Please describe.**

I'm always frustrated when using GRBL. Don't get me wrong, it gets the job done. But it's also been idle for over 3 years, and people have built a lot of code for MPCNC machines (and 3D printers). To name a few: error recovery, bed leveling with advanced sensors, VFD spindles using RS485, better handling (more pins) of endstops (which one is hit? where?), S-splines for steppers, and so on, and so on. After much consideration, I decided to give Marlin a spin that supports all these features.

Marlin is a nice piece of code, has a very active code/user base, has tons and tons of features and supported hardware, and having an Arduino DUE is awesome for running a CNC router because of the speed, buffering, etc. 

There's just one problem: Candle is the most awesome software for running a router, and it's build for GRBL, not Marlin.

**Describe the solution you'd like**

I would like Candle to support Marlin. It should be more or less the same, as Marlin also basically interprets G-Code. The main difference in the protocol is response handling and error handling.

**Describe alternatives you've considered**

CNCJS. The only alternative as of now I guess.

sinfocomp, 2020-06-29:
I to Love candle, but, lasergrbl supports marlin, you can also try grbl
mega, por grbl lpc, personally i use grbl mega

atlaste, 2020-06-29:
Yea I'm aware of the different variants of GRBL, frankly I believe GRBL_ESP32 is the best GRBL variant currently... but still, the reality is that Marlin has roughly 10 MB of source code now with dozens of active supporters, while GRBL is just a few KB's with a few forks being active. And if you want to add f.ex. a BLTouch for bed leveling, it's not even remotely funny to implement that in GRBL.

None of the branches of grbl runs on a due currently. None of them support my RS485 VFD spindle. Yes, I could use an analog / PWM signal instead, but that's a very poor solution as error handling goes entirely out of the window with that.

I love candle for multiple reasons. I run candle on my cnc pc, which runs linux. And while I like to work in c# myself, afaik lasergrbl doesn't support linux for a lot of c# related reasons.