Denvi/Candle #406: runtime error Большие.nc

ToJl9TopTonop, 2020-07-24:
Столкнулся с проблемой runtime error при загрузке больших .nc файлов, в версии 1.2.9.1b.
ошибка появляется в момент визуализации. если закрыть окно визуализации то все работает.
в версии 1.1.7 подобных проблем нет.
прикреплю пример .nc файла.
[пример.zip](https://github.com/Denvi/Candle/files/4973159/default.zip)

Denvi, 2020-07-31:
Попробуйте версию x64: [Candle_1.2.9.2b_x64.zip](https://github.com/Denvi/Candle/releases/download/v1.2b/Candle_1.2.9.2b_x64.zip)
Если нужна рабочая версия под x86, напишите, нужно будет другим компилятором собирать.

ToJl9TopTonop, 2020-08-09:
Замечательно работает! :) Пожалуй единственная программа что может тянуть такие файлы.

closed 2022-05-07 by Denvi