Denvi/Candle #408: Running the 1.2.9.2b version breaks version 1.1.7

nickm324, 2020-08-12:
**Versions**
 - Candle version: [e.g.: 1.0.11, 1.1.7, 1.2.9b]
 - Firmware version: [1.1g]

**PC info**
- OS: Windows 10
- OpenGL version: ...

**Describe the bug**
After testing version 1.2.9.2b_X64 and then attempting to run version 1.1.7 again fails.
When you double click the executable nothing happens, no error, just never starts. 
Assuming its realated to some registry entry that was created from 1.2.9.2b that 1.1.7 is trying to read and cannot due to new changes.

**To Reproduce**
Steps to reproduce the behavior:
1. Run 1.1.7 and exit.
2. Then run 1.2.9.2b and exit.
3. Go back and try and run 1.1.7 and it fails to start.

**Expected behavior**
Both version would still run on the same machine.

**Screenshots**

**Example files**

**Additional context**

ondras12345, 2020-08-12:
Make sure the programs are not attempting to use the same settings.ini file. Try deleting it.
I don't think Candle uses the Windows registry.

It works just fine on my computer (Windows 10 64bit 1909) - I can run 1.0.11, 1.1.7, 1.2.9.1b and 1.2.9.2b without any problems.

nickm324, 2020-08-12:
I found that the issues was actually due to me having a dual monitor setup and I opened it on my second monitor first and then exited. once I editing  the settings INI and removed the formGeometry line it then again worked and I opened it only on my primary screen going forward. Definitely a bug in the software. The fact that I tried both versions was just a coincidence.

closed 2020-08-12 by nickm324