Denvi/Candle #409: Text rendered as black squares

oomek, 2020-08-13:
**Versions**
 - Candle version: 1.2.9b
 - Firmware version: 1.1h

**PC info**
- OS: Windows 10 Home
- OpenGL version: 4.6.13596

**Describe the bug**
The latest version renders text as black squares, previous version renders text normally

**Screenshots**
[image]

Jose-Manuel34, 2020-12-18:
Hi, did you solve your problem. I have a similar situation and I haven't found a way to solve it.

[image]

oomek, 2020-12-18:
The issue resolved by itself a while back, but honestly I have no idea how. Maybe by the gpu driver update, who knows.

closed 2022-05-07 by Denvi