Denvi/Candle #412: Jog with micropositiong 0.001mm

AugustoBRAZIL, 2020-08-17:
Dear, you could include a manual jog smaller than the current one, I need 0.001mm micropositions. Would it be possible to include this 0.001mm button in the software?

ondras12345, 2020-08-17:
There should be `jogSteps=Continuously, 0.01, 0.1, 1, 5, 10, 100` in your settings.ini file. Try adding `0.001`.

AugustoBRAZIL, 2020-08-18:
Tks for the answer i try adding 0.001mm, 
it only activates when the 0.001mm step is pressed 4 times, ie the software responds to a minimum step of 0.004mm

ondras12345, 2020-08-18:
https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L3896
```
sendCommand(QString("$J=G21G91X%1Y%2Z%3F%4")
                    .arg(vec.x(), 0, 'g', 4)
                    .arg(vec.y(), 0, 'g', 4)
                    .arg(vec.z(), 0, 'g', 4)
                    .arg(speed), -3, m_settings->showUICommands());
```

The coordinates seem to be limited to 4 significant digits due to incorrect use of `'g'` instead of `'f'`, which could also cause problems with scientific notation being used for large numbers.

https://doc.qt.io/qt-5/qstring.html#argument-formats
> A precision is also specified with the argument format.
> For the 'e', 'E', and 'f' formats, the precision represents the number of digits **after the decimal point**.
> For the 'g' and 'G' formats, the precision represents the **maximum number of significant digits** (trailing zeroes are omitted).

Please try jogging closer to the origin (0,0,0), so that the coordinates are lower that 10 (number of significant figures <= 4).

If I'm right, I could potentially fix this and file a PR against the experimental branch.
However, you would need to compile the program yourself or wait for @Denvi to merge the PR and release a new version of the software.

ondras12345, 2020-08-18:
Does it work if you do something like this?
```
G91
G0 X0.001
(repeat G0 X0.001 for multiple steps)
G90
```

If not, please post your Grbl version and settings as it may not be a Candle-related problem.

AugustoBRAZIL, 2020-08-18:
dear i try G0 X0.001 and it doesn't work it only works with G0 X0.004. Could you fix this problem or contact @Denvi to fix the program? My version of Grbl is 1.1f and I am using the Candle_1.2.9.2b version

ondras12345, 2020-08-18:
> i try G0 X0.001 and it doesn't work it only works with G0 X0.004

And `G0 X0.001` four times? This seems to be a Grbl issue. Post your Grbl settings (`$$`).

AugustoBRAZIL, 2020-08-18:
doesn't work G0 X0.001 four times
My Grbl setting is:
$0=10; $1=25; $2=0; $3=0; $4=0; $5=0; $6=0; $10=1; $11=0.010; $12=0.002; $13=0; $20=0; $21=0; $22=0; $23=0; $24=25.000; $25=500.000; $26=250; $27=1.000; $30=1000; $31=0; $32=0; $100=250.000; $101=250.000; $102=250.000; $110=500.000; $111=500.000; $112=500.000; $120=10.000; $121=10.000; $122=10.000; $130=200.000; $131=200.000; $132=200.000; ok

ondras12345, 2020-08-18:
You only have 250 steps/mm ([`$100`](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#---view-grbl-settings)), 1/250=0.004

AugustoBRAZIL, 2020-08-18:
tks Dear my system have $ 100 = 3200 steps/mm, 
I got micropositions with 0.001mm, another question about the software, how would I be able to zero the work coordinates of the X axis and the Y axis separately only with a button for each axis (one button for zero X axis and one button for Y axis) ?

ondras12345, 2020-08-18:
The "Zero XY" buttons executes `G92X0Y0`, you can use `G92X0` to zero only the X axis.
You can assign the command to one of the user command buttons if you plan to use it regularly.
https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L2383
http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g92

AugustoBRAZIL, 2020-08-18:
Tks Dear, I put the function G92 X0 and G92 Y0 on the USERS COMANDS buttons