Denvi/Candle #415: Camera

BSD31, 2020-08-27:
Hello.
I am using a laptop for Candle and the internal camera is used by default by the camera plugin.
So I see myself very well but the interest is limited. Lol.
I don't know how to configure the camera's USB port so that Candle sees my current engraving coming from my CNC's camera, which would be much more interesting.
If not, is it possible to use an http video stream from my local network because I also have a WiFi camera which could very well do the job and that would eliminate a cable ?
Thanks for this work.
Bruno

ondras12345, 2020-08-27:
> I don't know how to configure the camera's USB port so that Candle sees my current engraving coming from my CNC's camera, which would be much more interesting.

Service > Settings > Camera > Name
https://github.com/Denvi/Candle/blob/experimental/src/designerplugins/cameraplugin/camerawidget.cpp#L42
https://doc.qt.io/qt-5/qcamerainfo.html#description

Use the camera name shown in Windows device manager.

giecuk, 2020-08-31:
> Service > Settings > Camera > Name
> https://github.com/Denvi/Candle/blob/experimental/src/designerplugins/cameraplugin/camerawidget.cpp#L42
> https://doc.qt.io/qt-5/qcamerainfo.html#description
>
> Use [...]

hii, i still confused...
could you please make a complete tutorial for it....
thanks.

ondras12345, 2020-08-31:
1. Open Windows Device Manager.
1. Find your camera.
1. Enter the name of the camera you see in Device Manager to Service > Settings > Camera > Name in Candle.

giecuk, 2020-08-31:
> 1. Open Windows Device Manager.
> 2. Find your camera.
> 3. Enter the name of the camera you see in Device Manager to Service > Settings > Camera > Name in Candle.

i tried it, but still failed...
in device manager my camera name is: USB2.0 PC CAMERA
it's i98 camera, visual ear cleaner.
when i try to test the camera using OBS Studio software, it works good.

BSD31, 2020-09-01:
Hello.
The proposed solution works perfectly for me.
A good evolution of Candle would be to be able to access our cameras via an http stream.
Thank you for this fabulous job.
Bruno from France

wayneosdias, 2020-09-09:
I am having a similar issue. I can get my integrated camera to work within Candle, but not a USB attached camera. The laptop Im using for this also controls a pick and place machine that utilizes x2 vision cameras and I have no issues. The 'USB Camera' I am trying to use works just fine within Win10 Camera app, but doesnt seem to work w Candle. Is there and alias beside the Display  Name within the device properties I can try?

Can others specify what camera models they have gotten to work with Candle?

Love this app :)

civanahmetyasin, 2020-09-20:
[image]
[image]

Like this

wayneosdias, 2020-09-20:
Yes, I understand. No it does not work...

picspb, 2020-09-28:
You need to configure privacy settings for Windows 10. Allow all applications to access the camera.
[image]

wayneosdias, 2020-09-28:
Ill try the privacy setting. I had to do something similar to get my pick
and place machine vision detection to work...

wayneosdias, 2020-09-29:
Please advise, I love this App and want to use it for another project!
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]

wayneosdias, 2020-09-29:
[https://drive.google.com/file/d/1hG3G9wav1XKo9tpFI6aVn7DIS6fSm_aD/view?usp=sharing](url)
[https://drive.google.com/file/d/1JWCESbrAIue5neZ23QpVSHfLiUKgrwQg/view?usp=sharing](url)
[https://drive.google.com/file/d/1qpv-Lq6HD5RCFNRdnrwVHH6HtZbgU8OL/view?usp=sharing](url)
[https://drive.google.com/file/d/1eENgI7TAEwOkg4E9hXijhdz6Kt0-_YTr/view?usp=sharing](url)
[https://drive.google.com/file/d/1yVBQCKMWiW7hXRQD2pWH5-WCyzONulvd/view?usp=sharing](url)

Wayne2580, 2020-11-30:
Still not working with all of the above suggestions. Ive also disabled/removed the integrated camera from the device manager and it still does not work. Ive heard that some people using openCV have issues using the device name. What seems to work better is using the device number within the cue.
[image]
[image]
[image]
[image]
[image]

PCA54, 2020-12-04:
Exactly the same issue that @Wayne2580 details. Works fine with integrated camera (name "HD WebCam") . does'nt work with USB camera (name "GENERAL - UVC")....  The laptop is running Windows 8.1
Tried all the suggestions listed above.
The camera works fine with other softwares (including bCNC)
Any idea welcome !!

PCA54, 2020-12-07:
Issue resolved.
The problem was with the windows name of the USB cam which has a trailing whitespace... ie "GENERAL - UVC "  and NOT "GENERAL - UVC"

Wayne2580, 2020-12-07:
Glad that helped you PCA54. This does NOT fix my issue. Ive tried this w 3+ different cameras and know to trim the white space. Its a SW problem as I use this laptop w my pick and place machine w dual vision. That SW works fine w any camera I use. I really like this program, I just wish I could get vision to work.

Wayne2580, 2021-01-26:
YES! I finally found the issue to my USB camera problem. The camera works plug and play fine in all other apps, but not Candle. Turns out it was the aspect ratio. I don't know recall what the default ratio in Candle is, 1280x720 or 1280x800 
[image]
I believe. Regardless, using the default ratio setting I got a black/blank screen. After changing the ratio to the available ratios listed in VLC and media player it works just fine. All the listed res/ratios listed within the screenshot work. Thanks for everyone's help on this, I really like this app :)

Wayne2580, 2021-01-26:
I just confirmed this makes the multiple cameras works within Candle on both my desktop and laptop. Just for clarification I can verify that the following cameras work within Candle/Win10 with default windows drivers if the correct res/ratio is edited with Service>Camera>Resolution.
https://www.amazon.com/gp/product/B085XZVFGT/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
[image]
https://www.amazon.com/gp/product/B07C93B564/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
[image]
I had another USB camera that didn't work initially, but I'm unable to locate at the moment to verify.
I still don't understand what the 'Position' property is, XY Offset from the tool?

closed 2021-12-15 by BSD31