Denvi/Candle #416: 1.2b Source is not up to date

nickm324, 2020-08-31:
**Versions**
 - Candle version: [1.2.9b]
 - Firmware version: [1.1h]

**PC info**
- OS: …
- OpenGL version: ...

**Describe the bug**
The version of source on the releases page for 1.2b is not the correct version that matches the 2 builds that are there.
Neither the ZIP or TAR source contains the current version of the code that includes the plugins, shortcuts or panel movement code.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the ZIP or TAR and compile.

**Expected behavior**
Expected the source to be of the current version of candle for 1.2b with the panels and webcam plugin,etc.

ondras12345, 2020-09-01:
The v1.2 tag from which github automatically generated the source code zip files points to one of the first commits of the experimental branch, so it does not include all the features.
However, v1.2 is still a pre-release, so I don't think this is a big problem.
@Denvi should definitely fix this when he releases it, though.
I would also like to see tags corresponding to all the beta builds (e.g. v1.2.9.2b).
In the meantime, you can build it from the latest commit of the experimental branch if you want the features.

nickm324, 2020-09-01:
Ok I will give that a try.

nickm324, 2020-09-03:
There is no source code for the 1.2.9b experimental branch. I have tried Clone/Download which only gives me the 1.1.8 code, the ZIP and TAR files on the release page for 1.2.9b. None of them have the source for the current version of the pre-built binary that is under the 1.2.9b release page.

ondras12345, 2020-09-03:
> I have tried Clone/Download which only gives me the 1.1.8 code

I _think_ the version number was not updated in the repo, but the source code contains the new features. (Unless you clicked the Download button while still at the master branch.)

nickm324, 2020-09-03:
> > I have tried Clone/Download which only gives me the 1.1.8 code
> 
> I _think_ the version number was not updated in the repo, but the source code contains the new features. (Unless you clicked the Download button while still at the master branch.)

I have tried all versions of the source code and NONE of them have the current updated source for 1.2.9b.
I have downloaded every version of source available here and built them each using QT and none of those have the same features as the BUILT version for download on the resources page for 1.2.

nickm324, 2020-09-21:
Can anyone tell me where to get the source for the 1.2b version, what is on the Releases page is NOT the correct source code, at least its not the same source that the built version was made from.

Denvi, 2020-09-21:
Hello, here: https://github.com/Denvi/Candle/tree/Experimental

nickm324, 2020-09-21:
Thanks I will check it out.