Denvi/Candle #423: Hint For Creating Macros

arinurcahyo, 2020-09-06:
Dear, 

I work with pvc colored plastic to make a nameplate
to zero probe the tools, I used a piece of 3mm thick cooper, and do these:
1. put in on the surface of material, 
2. z-probe it, 
3. then Zero Z it, 
4. then Safe Position
5. Run command G0Z-3
4. then Zero Z it (again)

Is there any command or macro to make it short? or can we set zero z to certain position other than the tool real position (for example, the tool position is 0,0,0 and we want to set zero z to 0,0,-3.

Regards,

Ari

ondras12345, 2020-09-06:
http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g92
`G92 Z3` will make the current point 0,0,3.

Copy your probe command to one of the user commands and append `; G92 Z3`. It will probe and automatically subtract the thickness of the copper plate.

arinurcahyo, 2020-09-07:
Thank You. It works perfectly.

I added safe position code at the beginning and the end of the code, so I can remove the copper plate immediately.

`G21G90; G53G0Z0; 
G91 Z-15; 
G21G91G38.2Z-30F100; G0Z1; G38.2Z-2F10; 
G92 Z3;
G21G90; G53G0Z0`

closed 2020-09-07 by arinurcahyo