Denvi/Candle #424: CNC 3018 PRO and ERROR: 2

Dawaherjesus, 2020-09-06:
- Candle version: [1.1.7]
 - Firmware version: [1.1f]
- OS: Windows 7 Professional Service Pack 1 64 Bits
I am modifying my Gerber file with FlatCAM program, I generate my .nc file and open it with Candle.
running the test mode randomly stops the process. stops with ERROR: 2.
I have already checked the code and everything is perfect.
I have tested the direct code on the CNC using the SD memory and it works perfectly.

[image]
[Error2.zip](https://github.com/Denvi/Candle/files/5180086/Error2.zip)

Denvi, 2020-09-07:
Looks like connection problem. Try different USB cable.

Dawaherjesus, 2020-09-07:
I have already tried another cable and the problem continues.

it seems like a problem with the flatcam.
What other gerber converter do you recommend?

Dawaherjesus, 2020-09-07:
I tell you, I tried to update the firmware of the CNC and it gave me Update Failed.
Dust off an old laptop, and started working with the laptop, everything settled. everything works perfectly.

Solution for Firmware Update Failed:
uninstall antivirus and disable Windows Defender.
I followed the update manual.

In the case of "Error: 2" of the Candle:
I don't know if the steps performed with the firmware update have something to do with it.
Another thing, the PC is in Spanish and the Laptop in English. is this the cause?

ondras12345, 2020-09-07:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages
Error:2 is a Grbl error that means "Numeric value format is not valid or missing an expected value."

closed 2025-09-10 by github-actions[bot]