Denvi/Candle #428: Candle mirror...

Dzintars-li, 2020-09-18:
I create file inArtCam, coordinates is set in left bottom corner.
But in the Candle produced part is crooped upside down and rotate 180 degrees...
How to resolve this? Maybe somone know?

Denvi, 2020-09-20:
Hi, check #391

Dzintars-li, 2020-09-20:
Thank you Denvi!
I find this post leater and problem is solved. On 3018 machine Y axis moving table and this is the problem when you push the down button... Y- button moving table "up" and loooks that the table goes to wrong direction...
I am so sorry but I bought this machine yust 6 days ago...
One day for mounting parts together and 4 days for searching and learning Grbl, Candle and ArtCam...
First cut was 18.09 and... problems accure!
My machines zero point is set to left upper corner(NW).
May you post link where I can find information about  4 User Button setting?
I can't find this information on GitHub, maybe I searching in wrong place?
Some time ago I see video on You Tube, but forget add to my bookmarks...
Thank you again, there is so mutch information but I should grow up to reach this level!

Braunfeltd, 2020-09-22:
you a Sainsmart member on FB that would be the best place imo to get the info you need on the 3018.   This is more for just the candle software.  its not a bug with Candle its something you are doing that is incorrect and or your settings in the controller $$ values.  there are ways to invert coordinates through the settings, however more than likely its how you are generating the files from Artcam.

closed 2025-09-10 by github-actions[bot]