Denvi/Candle #429: VEVOR CNC 3018-PRO 3 Axis CNC Router Kit - Offline Controller not being seen in Win 10

Thracken13, 2020-09-22:
hey all, i am very new to this - and this may not be the venue to find assistance, but thought i would try.

I just got the above CNC Machine, and have it setup - however when i plug the Offline controller into my PC - the computer does not assign a Comport number to it, and instead treats it as if it were a flash drive.

because no COM port, I cannot connect it to Candle to try my had at using the machine. I looked for about 5 hours online and in YouTube videos, but could not find anything to look into for it.

just wondering if anyone else has had this issue before. TY for any info you might be able to impart :)

Braunfeltd, 2020-09-22:
Offline controller is called offline as its meant to be used with out a computer attached to the 3018.   Also some GBRL controllers do not allow both computer and offline to be connected at the same time as the offline uses serial port.

Thracken13, 2020-09-22:
but if i have to plug it into my PC only to load the info from Candle, and my PC doesnt assign it a com port - how can Candle use it?

jblb, 2020-11-09:
plug your computer on the control board not on the offline controller, and disconnect the off line controller
that's work for me