Denvi/Candle #435: empty menu

moje01konto, 2020-09-30:
On fedora linux I have trouble with menu
[image]
How turn on visibility menu? 
How changing to Polish language?

ondras12345, 2020-09-30:
> How changing to Polish language?

https://github.com/Denvi/Candle/issues/98
```
Hi, there is no language setting in "Candle", it is selected by system regional settings.
```
Polish is not supported.
https://github.com/Denvi/Candle/tree/master/src/translations

I'm not sure about the menu.
Did you compile Candle yourself? What version?
@Denvi should be able to help you.

Rolf-R, 2020-12-07:
I‘m experiencing this issue with the empty menus, too. Version is 1.1.7 running on up to date Arch 

May be related to:
`user@host ~% LANG=C candle
locale: "C"
style "Windows"
style "GTK+"
QGtkStyle could not resolve GTK. Make sure you have installed the proper libraries.`
There are reports with other programs having this issue related to this message. The solution there is to set QT_STYLE_OVERRIDE=gtk2 which does not work in here.

bevinhex, 2020-12-28:
Same here, still trying to find a workaround, I"m guessing if we were using linux's default QT libraries, issue might gets solved with the QT_STYLE_OVERRIDE approach

bevinhex, 2020-12-28:
Did some tests, tried to run windows version with wine, runs fine , but as you can see in the picture, menu shows up on different monitor (right), while program is running on left monitor
[image]

ByteHamster, 2021-01-10:
> So anyone with the problem, feel free to setup qtcreator with qt5, then build the application.

@bevinhex How did you do that? After fixing some imports, I am getting the error message `cannot find -lcustomwidgets` and am pretty much stuck :(

bevinhex, 2021-01-13:
> > So anyone with the problem, feel free to setup qtcreator with qt5, then build the application.
> 
> @bevinhex How did you do that? After fixing some imports, I am getting the error message `cannot find -lcustomwidgets` and am pretty much stuck :(

I am using debian, qtcreator was installed long time ago, I just opened the project with qtcreator, then build it, it was complaining some issues like yours, it means some libraries are missing, `sudo apt install libqt5*` (double tab there should list a bunch of library names) should do the trick

here are my project file , although I don't remember changing anything, just in case
```
#-------------------------------------------------
#
# Project created by QtCreator 2014-05-23T17:51:21
#
#-------------------------------------------------

QT       = core gui opengl serialport
greaterThan(QT_MAJOR_VERSION, 4): QT += widgets

win32: {
    QT += winextras
    DEFINES += WINDOWS
    QMAKE_LFLAGS += "-Wl,--large-address-aware"
    QMAKE_CXXFLAGS_DEBUG += -g3 -pg
    QMAKE_LFLAGS_DEBUG += -pg -lgmon
}

unix:!macx {
    DEFINES += UNIX #GL_GLEXT_PROTOTYPES
    QMAKE_LFLAGS += "-Wl,-rpath,\'\$$ORIGIN/libs\'"
}

contains(QT_CONFIG, opengles.) {
    warning("GL ES detected. VAO will be disabled.")
    DEFINES += GLES
    INSTALLS += target
    target.path = /home/pi
}

TARGET = Candle
TEMPLATE = app
VERSION = 1.1.8
RC_ICONS += images/candle.ico

DEFINES += sNan=\"65536\"
DEFINES += APP_VERSION=\\\"$$VERSION\\\"

TRANSLATIONS += translations/candle_en.ts translations/candle_ru.ts translations/candle_es.ts translations/candle_fr.ts translations/candle_pt.ts

SOURCES += main.cpp\
        frmmain.cpp \
    frmsettings.cpp \
    frmabout.cpp \
    drawers/gcodedrawer.cpp \
    drawers/heightmapborderdrawer.cpp \
    drawers/heightmapgriddrawer.cpp \
    drawers/heightmapinterpolationdrawer.cpp \
    drawers/origindrawer.cpp \
    drawers/shaderdrawable.cpp \
    drawers/tooldrawer.cpp \
    parser/arcproperties.cpp \
    parser/gcodeparser.cpp \
    parser/gcodepreprocessorutils.cpp \
    parser/gcodeviewparse.cpp \
    parser/linesegment.cpp \
    parser/pointsegment.cpp \
    tables/gcodetablemodel.cpp \
    tables/heightmaptablemodel.cpp \
    widgets/colorpicker.cpp \
    widgets/combobox.cpp \
    widgets/groupbox.cpp \
    widgets/scrollarea.cpp \
    widgets/styledtoolbutton.cpp \
    widgets/widget.cpp \
    widgets/glwidget.cpp \
    widgets/slider.cpp \
    widgets/sliderbox.cpp \
    drawers/selectiondrawer.cpp \
    widgets/comboboxkey.cpp

HEADERS  += frmmain.h \
    frmsettings.h \
    frmabout.h \
    drawers/gcodedrawer.h \
    drawers/heightmapborderdrawer.h \
    drawers/heightmapgriddrawer.h \
    drawers/heightmapinterpolationdrawer.h \
    drawers/origindrawer.h \
    drawers/shaderdrawable.h \
    drawers/tooldrawer.h \
    parser/arcproperties.h \
    parser/gcodeparser.h \
    parser/gcodepreprocessorutils.h \
    parser/gcodeviewparse.h \
    parser/linesegment.h \
    parser/pointsegment.h \
    tables/gcodetablemodel.h \
    tables/heightmaptablemodel.h \
    utils/interpolation.h \
    utils/util.h \
    widgets/colorpicker.h \
    widgets/combobox.h \
    widgets/groupbox.h \
    widgets/scrollarea.h \
    widgets/styledtoolbutton.h \
    widgets/widget.h \
    widgets/glwidget.h \
    widgets/slider.h \
    widgets/sliderbox.h \
    drawers/selectiondrawer.h \
    widgets/comboboxkey.h

FORMS    += frmmain.ui \
    frmsettings.ui \
    frmabout.ui \
    widgets/sliderbox.ui

DEFINES += _USE_MATH_DEFINES

RESOURCES += \
    shaders.qrc \
    images.qrc

CONFIG += c++11
```

and I am building from `master` branch

ByteHamster, 2021-01-14:
@bevinhex Thanks a lot. I was on the `Experimental` branch - building `master` does succeed for me, too. Sorry, I somehow did not think about branches when writing this.

Anyway, @Denvi has now pushed commit 4d54131f04c56efb578114f6d80067d43bca4298 which guides me into the right direction :) Still have not managed to build it but it is making progress.

ByteHamster, 2021-01-14:
It is not really clean but I finally managed to build Candle on the `Experimental` branch on Arch Linux.

Some hints, in case anyone is interested:

- Import into qt creator
- If you get redefinition errors, edit the corresponding module's `scriptbindings/qtbindings/qtscript_*/*.pro` file to **not** include `com_trolltech_qt_custom`
- If you get errors linking `customwidgets`, modify the module's `scriptbindings/qtbindings/qtscript_*/*.pro` file to point to the right path (I modified `LIBS += -L` to use the absolute path of `designerplugins/customwidgetsplugin`, not including the `release` folder which was not generated for me)

Really nice! Feels like a completely new program with all those flexible panels :)

[image]

klembaris, 2021-01-14:
Hi, 
i can't manage to build Candle on the Experimental branch on Windows 10. Got this: 
Project ERROR: Unknown module(s) in QT: uiplugin

ByteHamster, 2021-01-19:
> Project ERROR: Unknown module(s) in QT: uiplugin

Did you import the top-level project file? To be honest, I don't know much about QT. If you just want to try the experimental version, you can also use the pre-release which has Windows binaries attached to it: https://github.com/Denvi/Candle/releases/tag/v1.2b

19504643, 2021-03-03:
> It is not really clean but I finally managed to build Candle on the `Experimental` branch on Arch Linux.
> 
> Some hints, in case anyone is interested:
> 
> * Import into qt creator
> * If you get redefinition errors, edit the corresponding module's `scriptbindings/qtbindings/qtscript_*/*.pro` file to **not** include `com_trolltech_qt_custom`
> * If you get errors linking `customwidgets`, modify the module's `scriptbindings/qtbindings/qtscript_*/*.pro` file to point to the right path (I modified `LIBS += -L` to use the absolute path of `designerplugins/customwidgetsplugin` containing the generated `.so` file, not including the `release` folder which was not generated for me)
> 
> Really nice! Feels like a completely new program with all those flexible panels :)
> 
> [image]

@Denvi  @ByteHamster 
Hi, finally I get it running on ubuntu 1804.5, but there are something wrong
1. Visualizer is black and I can not get the view
2. no camera window
[image]

ondras12345, 2021-03-03:
>     1. Visualizer is black and I can not get the view

Try resetting to default settings.

>     2. no camera window

I _think_ you need to build the plugins separately. I never tried that.

Also, you don't need qtcreator. To build on Ubuntu 20.04 (a few months ago, commit 72a573380dd79db07c2f9a33ca3e028268f8e4f0), I needed to get these packages:
```sh
sudo apt install qt5-default libqt5serialport5-dev qttools5-dev-tools
# the above is sufficient for master, Experimental also requires:
sudo apt install qtmultimedia5-dev qttools5-dev qtscript5-dev
```
Then just
```sh
qmake candle.pro
make 
```

19504643, 2021-03-04:
> > ```
> > 1. Visualizer is black and I can not get the view
> > ```
> 
> Try resetting to default settings.
> 
> > ```
> > 2. no camera window
> > ```
> 
> I _think_ you need to build the plugins separately. I never tried that.
> 
> Also, you don't need qtcreator. To build on Ubuntu 20.04 (a few months ago, commit [72a5733](https://github.com/Denvi/Candle/commit/72a573380dd79db07c2f9a33ca3e028268f8e4f0)), I needed to get these packages:
> 
> ```shell
> sudo apt install qt5-default libqt5serialport5-dev qttools5-dev-tools
> # the above is sufficient for master, Experimental also requires:
> sudo apt install qtmultimedia5-dev qttools5-dev qtscript5-dev
> ```
> 
> Then just
> 
> ```shell
> qmake candle.pro
> make 
> ```

how to build the plugins separately? open candleplugins.pro and build, then build candle.pro, it still has no camera window

scanff, 2021-05-15:
> how to build the plugins separately? open candleplugins.pro and build, then build candle.pro, it still has no camera window

The release does not show the menu text, I tweaked the QT5 settings on Linux Mint but nothing.  So I built the master as described and  everything compiles and links okay but I don't see the out  binary to run, Is it some place other than the src dir?

fandres, 2021-06-19:
Hello, I have the same problem using Debian 10 and Lubuntu 20.04. I guess it is due to some dependence but I do not know which. Someone manage to solve the problem?

ondras12345, 2021-06-19:
I was recently able to build the latest experimental version, including plugins, from source in Debian.
build https://github.com/Denvi/Candle/issues/488#issuecomment-858985048
and plugins https://github.com/Denvi/Candle/issues/488#issuecomment-859758961

daramos, 2021-09-26:
In my case - the issue was related to the release package of Candle being compiled for x86 instead of x64. `sudo apt-get install libgtk2.0-:i386` resolved the issue for me. 

I also had to run `sudo apt-get install libsm6:i386` to fix the `This application failed to start because it could not find or load the Qt platform plugin "xcb".` error referenced in #299.

wosym, 2024-02-13:
I see this on the latest stable build. Have installed all packages that were mentioned above here and even more. Problem persists. Any clues?

wosym, 2024-02-13:
Interesting... I had the same problem when using the release-binary **and** when building from source. However... after creating a docker container with the same OS Ubuntu, I managed to build and run it just fine. No error, and the menu items show up.