Denvi/Candle #436: CNC 3018 Prover SainSmart ?

REVESETOILES, 2020-10-03:
Hello,

I just acquired a small CNC 3018 Prover from SainSmart, I have two concerns. Only for single and double-sided PCBs.

The first: When I engrave a pcb and reposition the machine to the zero of X and Y by clicking on the original restitution for milling change in order to make the holes. The machine does not position itself in the same place.This shifts the holes of the tracks. I checked the possible games in the axes. they seem correct.
I contacted the manufacturer, he advises me to mark the zero point on the plate in order to correctly reposition the axes manually.

It is not top, because imprecise, Moreover not ideal to realize double faces.

Second concern: When I write a PCb, the first one is correct, but the following ones with the same settings remove all tracks. Do I have to reset the machine and the soft (Candle) with each new engraving?

Do you have any advice for me?
Thanks for your help.
Sincerely.

Rondore, 2020-10-25:
I can speak for the first issue. I have a ProVER as well and the thing does not home consistently; you have to zero it manually every time you put something on the bed. As for the second issue, I am not sure what you mean by track as I have not done any PCB engraving, but you should not need to reset the machine between engravings. I swap material out in my 3018, do another manual zero and run the next job. Maybe someone else can shed better light on the subject.

Braunfeltd, 2020-10-25:
step 1.  home machine
step 2. move to your Zero point.  write down machine coordinates for x y

on finish of first job home machine
in console type g53 x -......   or g53 y -.....   (.... means your machine values.)
this will move your machine back to original machine coordinates and allow you to 0,0 with out all the work.

always home first as that will reset the machine cords for that the commands will get you back.

you will still have to zero z always.

closed 2025-09-10 by github-actions[bot]