Denvi/Candle #437: can you add an opportunity for TCP or UDP connection?

Jackvillian, 2020-10-07:
the application can connect only through a serial port, and this work, but sometimes I wanna work through wifi or ethernet. can you please add this option to the connection bar? it would be great and very useful.

ondras12345, 2020-10-07:
You can use third-party software to create a virtual serial port.

In GNU/Linux, something like this should work:
```
socat pty,link=$HOME/vmodem0,waitslave tcp:192.168.1.111:23 &
```

Jackvillian, 2020-10-08:
> socat pty,link=$HOME/vmodem0,waitslave tcp:192.168.1.111:23

i'm a macos user, this not worked for me

Jackvillian, 2020-10-11:
i disable csrutil after this i can create virtual serial port through socat but candle app dont see this port , app see only hardware like usb or bluetooth .

ondras12345, 2020-10-11:
You don't need to create the virtual tty in /dev, you can have it in your $HOME instead.
You should just be able to type the path to the tty in Candle settings. It may be missing from the dropdown, but it should work if you specify it manually.
I'll see if I can get to testing this on my GNU/Linux system later today.

closed 2025-09-10 by github-actions[bot]