Denvi/Candle #440: New graphical interface!

vineef, 2020-10-21:
Hello everyone.

I'm here to propose a new graphical interface for Candle. The Black Candle. 

I am grateful for the efforts of the developers in providing software with full possibility of customization. But as a CNC user, I prefer a software with a good graphical interface and only with the functions I need.

That said, I am proposing this graphical interface, as nice colors and everything positioned where it should.

What did you think?

[image]

civanahmetyasin, 2020-10-21:
I like it. I agree, every user should be free about colors. And the dark theme must added.

nickm324, 2020-10-21:
Nice I like it, how do we get it?

vineef, 2020-10-21:
> Nice I like it, how do we get it?

I don't know if I can post other people's projects with some modifications. I am waiting for the original developer to know his opinion. Thank you for your interest.

ondras12345, 2020-10-22:
> I don't know if I can post other people's projects with some modifications.

Candle is licensed under the GPL: https://github.com/Denvi/Candle/blob/master/LICENSE
You may distribute your modified version of this software, but you must preserve the license.
More info here: https://choosealicense.com/licenses/gpl-3.0/

nickm324, 2020-10-22:
Well its open source so dont see an issue.

Wayne2580, 2021-01-26:
I really like this too. I made a nextion display grbl sender that was black background gray font to spare the screen. This matches very well :)

closed 2025-09-10 by github-actions[bot]