Denvi/Candle #441: documenting heightmap + home/probe behaviour

arc12, 2020-10-24:
I would like to know when the heightmap adjustments are applied, specifically to understand how to combine with homing and probing.
For example: is it possible to build a heightmap which can be used either side of a tool change, with homing/probing to set working coordinates?
For example: would it be workable to create a microswitch-based heightmap probe for non-conductive surface probing, mounting this as a tool?

DEF7, 2021-01-04:
The heightmap is applied to the G-code coordinates. It actually will break up feeds to more smoothly conform to the heightmap, rather than simply offsetting the Z coordinate of each feed. The heightmapping procedure starts by probing the center of your G-code's XY area to use as Z0 from which all other probe points in the heightmap grid will be in terms of. This means that you probably want to zero your machine's Z by the same point.

I designed and built my own ultra low-profile always-on probe from scratch, because before I had to stick foil down ontop of my workpieces and alligator clip the cutter and the foil on the board, and it took way too much of my time. A microswitch will eventually work, but I don't think it will be reliably accurate insofar as repeatability.

[image]

[image]

This thing has saved me countless hours over the last few years and works perfectly every time. When I heightmap a board for v-carving, I simply probe the center of the board, set that as machine Z0, then setup my heightmapping and it does its thing. When I take the probe out and put the cutter in I just have to make sure I re-zero the Z axis on  there so that the v-bit is just barely touching the board surface. I always just eyeball it, but you can do it by just putting a piece of regular paper between the bit and the workpiece and jogging it down until the paper snags - and Z0 that point.

Good luck!

arc12, 2021-01-04:
Thanks @DEF7 .
Over the Christmas holiday I got to grips with this. The foil approach which I've seen some people recommend did not appeal to me either; a messy kludge. The procedure I adopted was as you use.

I did use a microswitch, an Omron D2HW, and fabricated a holder to fit the tool collet. Performing multiple z-probe cycles, gave repeatable results in 11/12 trials, with only one trial "off" by 0.001mm.

First run: perfect results! It seems to me that heightmapping is absolutely critical for getting good carving results from small machines like 3018s, as the depth of cut which the machine construction can sustain is limited.

Leaving this issue open as I intend to contribute notes to the wiki.