Denvi/Candle #443: candle heightmap not working

jgidhub, 2020-11-03:
I am trying to get heightmap working but mij spindle keeps dropping and is not respondong on my probe.
in the visualizer i get PS: P  when my probe is touching the pcb.
If i do a manual z probe zero with  the control button it works.
Is there a setting witch i missed.

greetings jasper

plopp, 2020-11-10:
I'm experiencing the same behavior

plopp, 2020-11-10:
Short answer - the way I understand it: The probe is only active within 1 mm above your origin.

I got the same crash you described since I was using the Z-probe but had my workspace origin further down (Z-direction) at the spoilboard. This made the heightmap procedure push the spindle down to Z=1 mm at a fast speed, crashing down into my Z-probe, since it expects to start probing at Z=1 mm. So to get around that and just start using the heightmap I zeroed my z-level on top of the Z-probe itself and now I can move that around with the spindle as it probes. It will not work without modification, the way I do it, because of the offset of the height of the Z-probe.

Last section of Part 1 in this [guide](https://docs.sainsmart.com/article/kj4xzak19j-how-to-utilize-height-mapping-in-candle) gave me the clue:

_"With X and Y zeroed, and Z Probed, the last step you will need to do is manually jog your Z axis downwards until the tip of your engraving bit or end-mill is less than 1 mm from the surface of your stock material. **DO NOT RE-Zero Z at this point**."_

portfedh, 2020-12-13:
Im also having issues with the heightmap. 

My Problem. 
When I run the "Probe" button in the heightmap settings the spindle goes through the grid lines but never moves the Z axis up and down to take measurements. 

I have changed the boder settings, the probe grid numbers the Zt and Zb numbers and the interpolation numbers but nothing seems to have an effect on this. ¿Does anyone know how to fix this?

The Settings I Used 
"Use Heightmap" checked, "Show border" checked, "Auto" Selected, "Show Grid" Checked, Interpolation Grid X:2 Y:2 checked. Probe grid: set to test every 10mm, Border X:0 Y:0, Zt: 1, Zb: -1. 

Using 3018Prover, ran a homing cycle, used Z-probe to set Z height. And set my XY coordinates at the lower left corner of my PCB. Jogged the Z axis to be above .5mm from the pcb at orgin.

jgidhub, 2020-12-13:
Solved,
Issue is gone with a fresh download of candle software no clue what the problem was.

closed 2020-12-13 by jgidhub