Denvi/Candle #444: Mac Candle does not for on Big Sur

ahartman, 2020-11-13:
The latestCandle version worked reliably on my Mac with Catalina for many months.
This morning, after Big Sur installation, it stopped working; there is no communication with the CNC mill.

Please fix asap was I really like using Candle on my Mac.

Regards, ahartman, belgium

vol4enok, 2020-12-01:
at this  [link](https://drive.google.com/drive/folders/1AGgX804tFUevi9AKxamSViYKNyABAJ_W?usp=sharing) you can download the assembly 1.2.9.1b. it works on my mac big sur

ahartman, 2020-12-25:
Dear vol4enok,

Thank you, this new version runs on my Big Sur as well. However, the Z-probe command does not work well. Half of the time, the Z-axis moves down, stops at touch but then does not goes back up again and the green 'Run' indicator stays lit. Can somebody take a look at that?

Thanks in advance. ahartman, Belgium

vol4enok, 2020-12-25:
your code does the following
G21 - go to the metric coordinate system
G91 - work in relative coordinates
G38.2Z-40F100 - scan to a depth of -40 mm at a speed of 100 mm / min
G0Z0.5 - after triggering the probe, raise the Z-axis by 0.5 mm
G38.2Z-40F50 - start scanning from a height of 0.5 mm to a depth of -40 mm. O_o we are working on relative coordinates. Why drop another -40mm if the workpiece is 0.5mm below the probe? I think that with this command, the machine goes into an accident, because it goes beyond the size of the Z-axis.
G92Z19.5 - G92 command sets the origin of work coordinates Z-axis. your code sets the origin of the working coordinates at 19.5mm from the probe point. (maybe you need it that way, I don't know)
if you use soft borders in GRBL then your Z-axis with your code should be at least 80 mm. I don't know what kind of machine you have. 
Here is my code that works on my machine. Z axis - 40mm
G21G91G38.2Z-30F50; G0Z1; G38.2Z-2F10; G92Z0; G53G0Z0
I start scanning whenever the axis is in the home position. That is, at the highest point
G92Z0 - sets work coordinates along the Z-axis to 0.
G53G0Z0 - simply returns the Z-axis to its home position.

ahartman, 2020-12-27:
Dear vol4enok,

Thank you for reviewing my probe command.
Having the Z38.2 twice is stupid and serves no purpose.
I do not start scanning from the top of the Z-axis; my Z-axis is 140mm, os scanning would take a long time for nothing. I manually jog my Z-axis down before starting the scan.

The 19.5 mm in the G923 command is the height of the Z-probe touch plate.

I wil start using your version adapted for touch plate height:
G21G91G38.2Z-30F50 G0Z1 G38.2Z-2F10 G92Z19.5 G53G0Z0

Regards, ahartman

Polyscript, 2021-03-07:
Hi guys, sorry for butting in. Hoping someone could help- my machine was fine before but now the X,Y and Z Axis only go in one direction unless I jog the X left, then X will go right but only the amount of steps it went left and no more. Same with the Y and Z axis. Can anyone help please? I've been at it for days and cannot make sense of it. When I send a code the Z probe goes up a little, the spindle turns very slow for 2-3 seconds and stops. I'm using G21G91G38.2Z-30F100; G0Z1; G38.2Z-2F10; G92Z14.3; G90; G0Z17 on my machine.

Any help will be greatly appreciated.

Many Thanks

ondras12345, 2021-03-07:
> now the X,Y and Z Axis only go in one direction unless I jog the X left, then X will go right but only the amount of steps it went left and no more.

Sounds like [soft limits](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean) to me. Check your Grbl settings.