Denvi/Candle #445: Why is Candle moving the decimal point from my g-code file?

gbach55, 2020-11-14:
**Versions**
 - Candle version: [e.g.: 1.0.11, 1.1.7, 1.2.9b]
 - Firmware version: [e.g.: 0.9j, 1.1h]

**PC info**
- OS: …
- OpenGL version: ...

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Example files**
Add G-code files that cause problems.

**Additional context**
Add any other context about the problem here.

gbach55, 2020-11-14:
My copy of Candle is moving my SainSmart 4030 .001" for a command to move 1".   How do I fix this?   I can't find anything within settings to get it back to full scale.

ondras12345, 2020-11-14:
Please fill in the issue template in your first post. We need more info about this, particularly the an example of the G-code command you are having problems with.

Denvi, 2020-11-15:
Try to add G20 command to the program beginning.

gbach55, 2020-11-16:
Not sure yet that the G20 code helped... but I did find that after I repositioned the stock it worked.  It seems I had it placed outside the machine limits.  Anyway... things are working as expected now.   I am going to have fun with this machine!

closed 2020-11-16 by gbach55