Denvi/Candle #446: Use Candle to Post-Process with Height Map

reynolds087, 2020-11-17:
**Is your feature request related to a problem? Please describe.**
I use bdring/GRBL_Esp32 and run my programs from the SD card. I do not have a persistent serial connection to the machine, so I cannot run programs from Candle, and thus I cannot use the height map feature.

**Describe the solution you'd like**
GRBL_Esp32 supports control over telnet. If Candle could create the height map over telnet, then save the processed file so I can put it on my SD card and run it locally on the machine.

Another option which would be slightly less convenient: 1) Document how to create a height map file manually. I can manually probe the PCB, create the file, import it to Candle, and then save the processed file with the height map integrated onto my SD card.

**Describe alternatives you've considered**
I thought about running a USB cable to the CNC controller, but I'll need a repeater, plus I would have to redesign my controller board to fit the usb cable and it's very difficult to make a PCB without the mill. I scrap a lot of copper clad trying to etch a new controller board already.

**Additional context**
Supporting telnet communication would make Candle more relevant to some of the other flavors of GRBL besides the Mega328. GRBL will continue to develop past USB and I think that being able to create the height maps over telnet would be a very useful feature that I haven't found anyone else supporting yet.

reynolds087, 2020-12-01:
Bump.

Just looking for confirmation either yes or no.

whoim2, 2021-06-10:
Its really need. Using telnet as serial and export corrected file.

closed 2025-09-10 by github-actions[bot]