Denvi/Candle #450: Candle crashes As soon as it starts

rebeltaz, 2020-11-21:
**Versions**
 - Candle version: 1.1.7, 1.2.9b
 - Firmware version: unsure

**PC info**
- OS: Windows 7 64-bit
- OpenGL version: 2.1

**Describe the bug**
I have tried both versions 1.17 and 1.29 both 32- and 64-bit varieties. The issue is the same. The program will load, but after a couple of seconds, it will just crash. Windows reports that the program has stopped and has no idea why.

TDHofstetter, 2021-01-10:
I have exactly the same problem. I don't get a splash screen or anything, just the error.

rebeltaz, 2021-01-10:
> I have exactly the same problem. I don't get a splash screen or anything, just the error.

I don't know if this helps you, but in my case it turned out that the computer I was using didn't support the graphics requirement of the program. I installed it on a different computer and it worked/s fine.

TDHofstetter, 2021-01-10:
It required a newer version of OpenGL than 2.1?

I'm at the point now where I've moved to a Linux box where I'm busy installing LinuxCNC instead because installing Candle on it got too hairy. Nearly all my machines are running Linux these days, and I don't feel very good about replacing my good Dells with Win10 barges just to run Candle. 

If I'm really forced to... I'll drive the darned machine from a BeagleBone Black.

XiaoShiZi, 2021-01-11:
Same issue here.
Starting Candle grblcontrol from the lower bar on the screen but nothing starts. 
Did an update this weekend and it worked well, even after a reboot. 
Now having the unworking condition.
A short windows error in the taskmanager but nothing on display. Hard to get into the problem. 
Searching for Candle.exe reveals nothing... Seems something is erasing the software...
[image]

closed 2025-10-05 by github-actions[bot]