Denvi/Candle #453: Candle software cannot control board axis movement with Arduino uno

lianzaozi, 2020-12-07:
I can use the candle software to connect to the Arduino uno control board, but the software cannot change the axis coordinates, and when I open GCode, the software does not display the motion track.

lianzaozi, 2020-12-07:
The order in which I operate the software

[image]

ondras12345, 2020-12-07:
Grbl version? Candle version? Grbl settings?

lianzaozi, 2020-12-07:
I can control it with other GUIs, but Candle software doesn't.

Grbl version: Candle_1.2.9.2b

Candle version: v1.1h (2019-08-25) Release

Grbl settings:
$$
< $0=10 (Sets time length per step. Minimum 3usec.)
< $1=25 (Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay.)
< $2=0 (Inverts the step signal. Set axis bit to invert (00000ZYX).)
< $3=0 (Inverts the direction signal. Set axis bit to invert (00000ZYX).)
< $4=0 (Inverts the stepper driver enable pin signal.)
< $5=0 (Inverts the all of the limit input pins.)
< $6=0 (Inverts the probe input pin signal.)
< $10=0 (Alters data included in status reports.)
< $11=0.010 (Sets how fast Grbl travels through consecutive motions. Lower value slows it down.)
< $12=0.002 (Sets the G2 and G3 arc tracing accuracy based on radial error. Beware: A very small value may effect performance.)
< $13=0 (Enables inch units when returning any position and rate value that is not a settings value.)
< $20=0 (Enables soft limits checks within machine travel and sets alarm when exceeded. Requires homing.)
< $21=0 (Enables hard limits. Immediately halts motion and throws an alarm when switch is triggered.)
< $22=0 (Enables homing cycle. Requires limit switches on all axes.)
< $23=0 (Homing searches for a switch in the positive direction. Set axis bit (00000ZYX) to search in negative direction.)
< $24=25.000 (Feed rate to slowly engage limit switch to determine its location accurately.)
< $25=500.000 (Seek rate to quickly find the limit switch before the slower locating phase.)
< $26=250 (Sets a short delay between phases of homing cycle to let a switch debounce.)
< $27=1.000 (Retract distance after triggering switch to disengage it. Homing will fail if switch isn't cleared.)
< $30=1000 (Maximum spindle speed. Sets PWM to 100% duty cycle.)
< $31=0 (Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.)
< $32=0 (Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.)
< $100=250.000 (X-axis travel resolution in steps per millimeter.)
< $101=250.000 (Y-axis travel resolution in steps per millimeter.)
< $102=250.000 (Z-axis travel resolution in steps per millimeter.)
< $110=500.000 (X-axis maximum rate. Used as G0 rapid rate.)
< $111=500.000 (Y-axis maximum rate. Used as G0 rapid rate.)
< $112=500.000 (Z-axis maximum rate. Used as G0 rapid rate.)
< $120=10.000 (X-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $121=10.000 (Y-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $122=10.000 (Z-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $130=200.000 (Maximum X-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $131=200.000 (Maximum Y-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $132=200.000 (Maximum Z-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< ok

ondras12345, 2020-12-07:
Try `$10=1`

lianzaozi, 2020-12-08:
tThank you. I'll test it.

lianzaozi, 2020-12-08:
Changed the setting of $10, and now the candle software can control it. But there is another problem, the candle software does not draw the track when I open GCode.

The red box in the screenshot below does not show the track
[image]

lianzaozi, 2020-12-08:
According to your instructions, the software is now available. Thank you

rezanasori, 2022-04-22:
ممنون و سپاسگزار برای سایت عالی شما