Denvi/Candle #456: Inch or mm

pelowitz, 2021-01-05:
Jogging seems to be hard coded to begin with G21 which forces mm. 

Although this is not uncommon what should happen is the jog command should follow the machine configuration.

if the machine reports in inches (GRBL $13=1) then jog $j should start with a G20 command and if the connected machine is configured for mm then jog should start with a G21.

I'm sure this OPINION is going to get flamed but if one is normally running configured in mm this would not present any change to what is currently being done.  But for those that prefer to work in inches jogging will work with the least amount of astonishment (generally a good thing when creating software).

ondras12345, 2021-01-05:
> Jogging seems to be hard coded to begin with G21 which forces mm.

https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L3886
Yes, it is.
However, this should [not affect the parser state](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging) and thus only applies to jogging, not the commands you run afterwards.

> jog command should follow the machine configuration

There is actually a Candle setting for this that is [used in other places](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L3065), so I think it should rather follow that.

> ... removing the hardcoded G21 in the jog command.

This seems like a really bad idea. If the user is expecting the machine to move in mm but the program ran previously set `G20`, he would be in for an unpleasant surprise.

Just looking at the source code, I don't see a simple way to change this behaviour without recompiling the program.

If you just want the steps and feeds in the jog menu to be in inches, it should be possible to fix this by using the [`double frmMain::toMetric(double value)`](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L3063) function and keeping `G21`. However, I guess the set of values available for the dropdowns should be different, too, so this may require a larger change. (The values can be changed manually in `settings.ini` if necessary, but the software should handle that.)
```cpp
void frmMain::jogStep()
{
    if (m_jogVector.length() == 0) return;

    if (ui->cboJogStep->currentText().toDouble() == 0) {
        const double acc = m_settings->acceleration();              // Acceleration mm/sec^2
        int speed = ui->cboJogFeed->currentText().toInt();          // Speed mm/min
        double v = (double)speed / 60;                              // Rapid speed mm/sec
        int N = 15;                                                 // Planner blocks
        double dt = qMax(0.01, sqrt(v) / (2 * acc * (N - 1)));      // Single jog command time
        double s = v * dt;                                          // Jog distance

        QVector3D vec = m_jogVector.normalized() * s;

    //    qDebug() << "jog" << speed << v << acc << dt <<s;

        sendCommand(QString("$J=G21G91X%1Y%2Z%3F%4")
                    .arg(vec.x(), 0, 'g', 4)
                    .arg(vec.y(), 0, 'g', 4)
                    .arg(vec.z(), 0, 'g', 4)
                    .arg(speed), -2, m_settings->showUICommands());
    } else {
        int speed = ui->cboJogFeed->currentText().toInt();          // Speed mm/min
        QVector3D vec = m_jogVector * ui->cboJogStep->currentText().toDouble();

        sendCommand(QString("$J=G21G91X%1Y%2Z%3F%4")
                    .arg(vec.x(), 0, 'g', 4)
                    .arg(vec.y(), 0, 'g', 4)
                    .arg(vec.z(), 0, 'g', 4)
                    .arg(speed), -3, m_settings->showUICommands());
    }
}
```
would become something like
```cpp
void frmMain::jogStep()
{
    if (m_jogVector.length() == 0) return;

    if (ui->cboJogStep->currentText().toDouble() == 0) {
        const double acc = m_settings->acceleration();              // Acceleration mm/sec^2
        int speed = toMetric(ui->cboJogFeed->currentText().toInt());// Speed mm/min CHANGE HERE
        double v = (double)speed / 60;                              // Rapid speed mm/sec
        int N = 15;                                                 // Planner blocks
        double dt = qMax(0.01, sqrt(v) / (2 * acc * (N - 1)));      // Single jog command time
        double s = v * dt;                                          // Jog distance

        QVector3D vec = m_jogVector.normalized() * s;

    //    qDebug() << "jog" << speed << v << acc << dt <<s;

        sendCommand(QString("$J=G21G91X%1Y%2Z%3F%4")
                    .arg(vec.x(), 0, 'g', 4)
                    .arg(vec.y(), 0, 'g', 4)
                    .arg(vec.z(), 0, 'g', 4)
                    .arg(speed), -2, m_settings->showUICommands());
    } else {
        int speed = toMetric(ui->cboJogFeed->currentText().toInt());// Speed mm/min CHANGE HERE
        QVector3D vec = m_jogVector * toMetric(ui->cboJogStep->currentText().toDouble()); // CHANGE HERE

        sendCommand(QString("$J=G21G91X%1Y%2Z%3F%4")
                    .arg(vec.x(), 0, 'g', 4)
                    .arg(vec.y(), 0, 'g', 4)
                    .arg(vec.z(), 0, 'g', 4)
                    .arg(speed), -3, m_settings->showUICommands());
    }
}
```
I have **NOT tested** this yet, but I may get to do that tomorrow and submit a PR if I get it working.

pelowitz, 2021-01-06:
As a quick fix for me as I rarely ever use metric I just edited the executable changing the G21 to G20 and noted that by changing the exe’s name to differentiate it from the original.  This works VERY well for me.

 

In my opinion following the machine configuration as reported in the $13 setting would be the best solution.  I would expect the $13 would be set deliberately because the user wants the machine to work in that mode.  In that case following the machine configuration mode would be the least astonishment to the user.

 

All of the other suggestions were intended to simply be possibilities.  I agree that removing the G21 is not a good idea for the reasons you mentioned.  Adding a button to allow the user to select just adds clutter to the ui.

ondras12345, 2021-01-06:
> In my opinion following the machine configuration as reported in the $13 setting would be the best solution.

Yes, but that's a completely separate issue. This "Imperial jogging" feature should be consistent with the rest of the SW, that is follow Candle's settings. This Candle setting (as well as "Acceleration" and "Rapid speed") should probably just follow the Grbl settings.

ondras12345, 2021-01-06:
> This Candle setting (as well as "Acceleration" and "Rapid speed") should probably just follow the Grbl settings.

Actually, it seems like the `Experimental` branch (at the time of writing, the latest commit is [`72a573380dd79db07c2f9a33ca3e028268f8e4f0`](https://github.com/Denvi/Candle/tree/72a573380dd79db07c2f9a33ca3e028268f8e4f0)) does just that:
https://github.com/Denvi/Candle/blob/72a573380dd79db07c2f9a33ca3e028268f8e4f0/src/candle/frmmain.cpp#L1328-L1342

Until today, I was actually unable to compile the SW from this branch (Ubuntu 64bit), but I figured it out. For future reference, these are the packages I needed to install:
```sh
sudo apt install qt5-default libqt5serialport5-dev qttools5-dev-tools
# the above is sufficient for master, Experimental also requires:
sudo apt install qtmultimedia5-dev qttools5-dev qtscript5-dev
```
I also needed to add these 2 lines to `src/candle/frmmain.cpp`:
```cpp
#include <QDrag>
#include <QTranslator>
```

It seems like the way metric vs imperial is handled in other functions has changed, too:
old
https://github.com/Denvi/Candle/blob/8fd07bfc7ffb6455936e926ba9bc38b4e714988a/src/frmmain.cpp#L2008-L2017
new
https://github.com/Denvi/Candle/blob/72a573380dd79db07c2f9a33ca3e028268f8e4f0/src/candle/frmmain.cpp#L2179-L2193

However, I think this was done just to simplify the code and my fix could remain pretty much the same.
I'll go and test it.

ondras12345, 2021-01-06:
Ok, it seems to be working. However, one issue I noticed is that if `$13` is changed while Candle is running, the Candle setting does not change until you restart it. This is not a problem with the code I added, Candle just [reads the Grbl settings](https://github.com/Denvi/Candle/blob/72a573380dd79db07c2f9a33ca3e028268f8e4f0/src/candle/frmmain.cpp#L1353) once (actually, it seems like it reads the settings each time a soft reset command `^X` is issued). I don't think this is a big issue, though.
I'll file a PR with the changes I made.

pelowitz, 2021-01-06:
Although your concern about changing $13 while running is a valid concern one should consider that typically, all of the $ settings are written to EEPROM which has a 100,000 cycle endurance.  That may seem like a lot but if a $ command is set in a g-code then every time it is run the EEPROM will be updated.  Doing this should be avoided if at all possible—it is a very poor practice.  Even changing $32 back and forth to change support (as is being done on the 3018 class of systems) between laser and spindle use is not a good idea.  A much better solution to switch between laser and spindle is to automatically wrap G0 moves with S0/Sxxx when using laser.  It is more complex than that as GRBL changes the interpretation of M3 and M4 depending on the state of $32.  

Either way thank you for your efforts on resolving this cgs/fps question.

ondras12345, 2021-01-06:
Yeah, I wasn't thinking about normal use all that much. I just wanted to prevent the "I just tested it and it still jogs in mm" comments...
I don't believe a hobbyist will ever kill the EEPROM in this application, but you're right - this is a potential problem.

ondras12345, 2021-03-23:
@pelowitz I *think* this should be fixed in 1.12.13b. Could you please try it and close this issue if it is fixed?

pelowitz, 2021-03-23:
I'll play with it.  Thanks.

closed 2021-03-23 by pelowitz