Denvi/Candle #457: CH341SER_MAC no longer visible as connection port

ahartman, 2021-01-06:
I am using Candle 1.2.9.1b and the CH34x driver, version 1.4 for a number of weeks on Big Sur.
From one minute to the next, the driver name has disappeared in the port connection dropdown in Candle preferences. I now only see tty.Bluetooth-Incoming-Port and cu. tty.Bluetooth-Incoming-Port and these do not work for me. So, I can no longer use Candle with my CNC mill.

On github "adrianmihalko/ch340g-ch34g-ch34x-mac-os-x-driver" I read that I should not install CH341serial at all as Apple supplies a CH34 driver from Mojave on. This is maybe the tty.Bluetooth-Incoming-Port but that does not work for me and the CH341 driver worked OK until 1 hour ago.

Any ideas?

Regards, ahartman, Belgium

ondras12345, 2021-01-06:
This might actually be completely unrelated to Candle.

Could you please show us the output of the following commands?
https://stackoverflow.com/questions/3815211/mac-command-line-list-available-serial-ports
```sh
ls /dev/tty.*
ls /dev/cu.*
```

ahartman, 2021-01-06:
Dear ondras12345

Please find attached; this is after having installed MAC341serial v1.5.

Regards, ahartman, belgium
[Terminal.txt](https://github.com/Denvi/Candle/files/5776443/Terminal.txt)

ondras12345, 2021-01-06:
Ok, your system does not see the serial port. This has nothing to do with Candle and may even be a hardware problem. CH340 is known to be unreliable. I have no experience with Macs, so I cannot help you with this. On a GNU/Linux operating system, I'd try to look at `dmesg` to see if this is a hardware problem or something driver-related.

ahartman, 2021-01-06:
Mac OSX is a certified Unix system under the hood, so I have dmesg. Can you see something in attached output?

Could you ask your collegue that is responsible for the Mac version of Candle how Candle could use the CH34 driver that Apple is said to supply in OSX Mojave and newer?

Thanks in advance.
[dmesg.txt](https://github.com/Denvi/Candle/files/5776578/dmesg.txt)

ondras12345, 2021-01-06:
I'm not a Candle dev, just a random user.
Candle does not need to "support" a driver. All the driver does is it communicates with the USB device and provides a standard serial port.

ondras12345, 2021-01-06:
Could you please try disconnecting and reconnecting the USB device while watching for new lines in `dmesg`?

closed 2025-09-09 by github-actions[bot]