Denvi/Candle #460: work area outline

M0USS, 2021-02-03:
I am trying to add the contour function of the work area, like in bcnc (see photo)

I put the code below:
G0 X [xmin] Y [ymin] G0 X [xmax] G0 Y [ymax] G0 X [xmin] G0 Y [ymin]
and he sends me back:
G0 X [XMIN] Y [YMIN] G0 X [XMAX] G0 Y [YMAX] G0 X [XMIN] G0 Y [YMIN] <error: 2

would you have a solution to my problem? maybe someone already made it?

Thank you in advance.

[image]
[image]

ondras12345, 2021-02-03:
> G0 X [XMIN] Y [YMIN] G0 X [XMAX] G0 Y [YMAX] G0 X [XMIN] G0 Y [YMIN] <error: 2

Are you really sending "[XMIN]" instead of the actual number? You need to provide Grbl with absolute coordinates.
> error:2: Numeric value format is not valid or missing an expected value.
> https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages

The beta version of Candle supports variables of some sort, but I haven't looked into that yet.

Also, I don't think you'll be able to send multiple commands in one line.
> error:21: More than one g-code command from same modal group found in block.

M0USS, 2021-02-03:
I don't know how to get the values, in order to put it between the brackets .. I don't know how Bcnc comes up with this Gcode! I will continue to search.

ondras12345, 2021-02-03:
I don't think it is possible to implement this feature at this time without modifying the source code of Candle.

M0USS, 2021-02-03:
Thanks, I am using Candle under Windows 10 so I cannot manipulate the code .. I will wait and see if the function will be added!

ondras12345, 2021-02-03:
https://github.com/Denvi/Candle/blob/75bebc5bcc6af37a26ba7bce4136dd63800a9b0a/src/candle/widgets/glwidget.cpp#L491
This seems to be the code that prints out the numbers shown in the bottom left corner of the visualizer. It has the variables you would need to implement this feature.

It should be possible to compile Candle from source in Windows, although I have never tried that. However, I think Denvi is doing all of the development on a Windows machine.

TheModernWizard, 2021-03-23:
Yes, it is possible to build and deploy Candle on Windows. I did it just yesterday after some trial and error (never compiled open source code before). However, there a few lines of code that need to be changed for it to compile. If anyone who reads this is interested, just post here and I will gladly do a write-up.

closed 2025-09-22 by Denvi