Denvi/Candle #461: untimely disconnections from the USB port

frorgaba, 2021-02-08:
**Versions**
 - Candle version:1.2.9b , 1.2.11b,  1.2.12.b
 - Firmware version:  1.1h

**PC info**
- OS: Windows 10 64bit
- OpenGL version:  4.6

when I run a milling operation on my cnc 3018, it often happens that the program loses the USB connection. 
this disconnection is more frequent on version 1.2.11.b and 1.2.12.b 
changing the USB port, restarting the program, the PC or the cnc do not solve the problem.

ondras12345, 2021-02-09:
When it disconnects, does your operating system still see the device (COM port)? If not, this is probably not related to Candle.

pedjas, 2021-03-16:
This is possibly due to RFI that is produced by spindle motor.
Sometimes just changing USB cable helps.

If that does not help, you have to try other methods of blocking RFI on your machine.

closed 2025-09-09 by github-actions[bot]