Denvi/Candle #471: Candle GRBL on Big Sur

Polyscript, 2021-03-07:
Hi guys hoping someone could help please- my machine was fine before but now the X,Y and Z Axis only go in one direction unless I jog the X left, it will go left, then when I choose right X will go right but only the amount of steps it went left and no more. Same with the Y and Z axis. Can anyone help please? I've been at it for days and cannot make sense of it. When I send a code the Z probe goes up a little, the spindle turns very slow for 2-3 seconds and stops. I'm using G21G91G38.2Z-30F100; G0Z1; G38.2Z-2F10; G92Z14.3; G90; G0Z17 on my machine.

Any help will be greatly appreciated.

I am using Candle 1.2.9.1b.dmg on the latest Big Sur update.

I apologise if this isn't the right format to ask the question- i'm new to all this.

Many Thanks

ondras12345, 2021-03-07:
> now the X,Y and Z Axis only go in one direction unless I jog the X left, then X will go right but only the amount of steps it went left and no more.

Sounds like [soft limits](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean) to me. Check your Grbl settings. We'll also need to know your Grbl version.

Polyscript, 2021-03-07:
> Sounds like [soft limits](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean) to me. Check your Grbl settings. We'll also need to know your Grbl version.

Hi, thank you for replying- I really appreciate it. I checked the I am running 1.2.9.1b of Candle and the settings are:

< $0=10
$1=25
$2=0
$3=2
$4=0
$5=0
$6=0
$10=0
$11=0.010
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=796.019
$101=794.702
$102=800.000
$110=2000.000
$111=2000.000
$112=600.000
$120=500.000
$121=500.000
$122=10.000
$130=260.000
$131=158.000
$132=34.000

my probe commands are G21G91G38.2Z-30F100; G0Z1; G38.2Z-2F10; G92Z14.3; G90; G0Z17

Polyscript, 2021-03-07:
> Do you have limit switches on your machine? If not, try `$20=0`; `$21=0`; `$22=0`.
> Please post your Grbl (machine firmware) version if this does not help. It is shown in the Console window when the machine starts.

You're awesome.... it's now moving!!!! thank you, thank you, thank you. I just need to get my code to start from the middle and i'm all good. Cheers buddy. I was at this for days!!

ondras12345, 2021-03-08:
>  just need to get my code to start from the middle

That won't be easy. Some G-Code commands change states of the parser, so you need to run the relevant commands before you continue from the line where you left off. Check out this [G-Code reference](http://linuxcnc.org/docs/html/gcode.html).

There seems to be a feature for this in Candle, although I've never tried it myself. Select the line you want to continue from, hold down the Send button and select Send from current line.

Do you think we can close this issue now?