Denvi/Candle #472: built success but no camera window

19504643, 2021-03-09:
**Versions**
 - Candle version: [Experimental]
 - Firmware version: [Experimental]

**PC info**
- OS: Ubuntu 18.04.5
- OpenGL version: built in Qt
- Also tried on Windows7

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
when built success, open the program, I can not find the camera window
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

yeyeto2788, 2022-02-04:
Have you tried restoring to the default settings? I have installed and saw the same but restoring settings to default worked for me. The problem is that you have all colors set to black.

I installed it using a [tool I created](https://github.com/yeyeto2788/cnc_setup_ansible) you can see the screenshot of it running.
[image]

wamad, 2023-07-10:
Hi @19504643 ,

Did you find the solution? Kindly share with me. 

Thank you.

closed 2025-09-09 by github-actions[bot]