Denvi/Candle #475: Heightmap Probing Not Working On Multi Axis Machine

Teselation, 2021-04-11:
**Versions**
 - Candle version: all versions
 - Firmware version: Grbl_Esp32
 
**PC info**
not relevant

**Describe the bug**
Probing button (inside Candle) works correctly.
When i tried to probe a Heightmap (6 axis setup) machine moved as usual (exactly like it would with Arduino) but the results on each probe (Heightmap results) were always zero independently from the height at which the probe detected the surface.

_It seems that the Candle takes first value from the right (unlike when its probing with the Z-probe button) of the returned coordinates. When i reduce number of axes to 3 Heightmap probing works as intended._

**To Reproduce**
Steps to reproduce the behavior:
1. Probe Heightmap with more than 3 axes set up inside Grbl_Esp32.
2. See machine moving correctly but candle reading 0.00 for each height.

**Expected behavior**
I expect it to read values correctly.

**Screenshots**
### Here is working sample from Arduino
[image]
[CTRL+X] < Grbl 1.1h ['$' for help]
G92Z0 < ok
$# < [G54:-102.000,-53.000,-90.200]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,90.200]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
G92X0Y0 < ok
$# < [G54:-102.000,-53.000,-90.200]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:102.000,53.000,90.200]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
G21G90F10G0Z1 < ok
G0X0Y0 < ok
G38.2Z-1 < [PRB:0.000,0.000,0.385:1]
ok
G0Z1 < ok
G0X0.000Y0.000 < ok
G38.2Z-1 < [PRB:0.000,0.000,0.623:1]
ok
G0Z1 < ok
G0X50.000Y0.000 < ok
G38.2Z-1 < [PRB:49.998,0.000,0.412:1]
ok
G0Z1 < ok
G0X100.000Y0.000 < ok
G38.2Z-1 < [PRB:100.000,0.000,-0.283:1]
ok
G0Z1 < ok
G0X100.000Y39.495 < ok
G38.2Z-1 < [PRB:100.000,39.495,0.030:1]
ok
G0Z1 < ok
G0X50.000Y39.495 < ok
G38.2Z-1 < [PRB:49.998,39.495,0.431:1]
ok
G0Z1 < ok
G0X0.000Y39.495 < ok
G38.2Z-1 < [PRB:0.000,39.495,0.370:1]
ok
G0Z1 < ok
G0X0.000Y78.990 < ok
G38.2Z-1 < [PRB:0.000,78.989,0.223:1]
ok
G0Z1 < ok
G0X50.000Y78.990 < ok
G38.2Z-1 < [PRB:49.998,78.989,0.586:1]
ok
G0Z1 < ok
G0X100.000Y78.990 < ok
G38.2Z-1 < [PRB:100.000,78.989,0.597:1]
ok
G0Z1 < ok

### Here is not working sample from Grbl_Esp32
[image]
[CTRL+X] < Grbl 1.3a ['$' for help]
S9600 < ok
G21G90F10G0Z1 < ok
G0X0Y0 < ok
G38.2Z-1 < [MSG:Found]
[PRB:0.000,0.000,-0.064,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X0.000Y0.000 < ok
G38.2Z-1 < [MSG:Found]
[PRB:0.000,0.000,1.000,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X50.000Y0.000 < ok
G38.2Z-1 < [MSG:Found]
[PRB:50.000,0.000,-0.139,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X100.000Y0.000 < ok
G38.2Z-1 < [MSG:Found]
[PRB:100.000,0.000,-0.331,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X100.000Y39.495 < ok
G38.2Z-1 < [MSG:Found]
[PRB:100.000,39.495,-0.207,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X50.000Y39.495 < ok
G38.2Z-1 < [MSG:Found]
[PRB:50.000,39.495,-0.145,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X0.000Y39.495 < ok
G38.2Z-1 < [MSG:Found]
[PRB:0.000,39.495,-0.181,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X0.000Y78.990 < ok
G38.2Z-1 < [MSG:Found]
[PRB:0.000,78.990,-0.201,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X50.000Y78.990 < ok
G38.2Z-1 < [MSG:Found]
[PRB:50.000,78.990,-0.083,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X100.000Y78.990 < ok
G38.2Z-1 < [MSG:Found]
[PRB:100.000,78.990,-0.034,0.000,0.000,0.000:1]
ok
G0Z1 < ok

### This is Grbl_Esp32 (set up for 6 axies) probe action (using Z-probe button) (working perfectly fine)
[image]

**Example files**
not needed

**Additional context**
Z-probe button works fine, Heightmap probing doesn't.

ondras12345, 2021-04-11:
Related https://github.com/Denvi/Candle/issues/277

Teselation, 2021-04-11:
Is it going to be fixed in newer versions of Candle?

closed 2021-04-12 by Teselation