Denvi/Candle #476: I Need Help

SWIN2121, 2021-04-19:
**Versions**
 - Candle version: [e.g.: 1.0.11, 1.1.7, 1.2.9b]
 - Firmware version: [e.g.: 0.9j, 1.1h]

**PC info**
- OS: …
- OpenGL version: ...

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Example files**
Add G-code files that cause problems.

**Additional context**
Add any other context about the problem here.

SWIN2121, 2021-04-19:
I have a cnc 3018 Pro my Z axis direction #391. I sat sat all my $ codes to Tipldn list posted on June 7 2020. I set my Home X, Y,Z on bottom Left. Now it is trying to cut but it goes to the Bottom and I have to Abort the cut. The reason that I did all of this is the job controler, even set on continusley and Feed at 2000 it was running really slow. How do I fix this mess. I would appreciate it for any help I can get. Thank You.

CNCaddicted, 2021-04-21:
Continuous and 2000 are jog feed/speed. The job will run at the feed rate in the G-code file. It may use the same feed rate for all movements or it may use multiple feed rates throughout the program. A simple way to check is to open the .NC file with Notepad or similar text editor and do a search for the letter "F" to see what they are. Which program generated the G-Code? Or did you hand code it?

CNCaddicted, 2021-04-21:
You may want to compare your settings with mine. I use homing/limit switches that home to the lower left hand corner.
[CTRL+X] < Grbl 1.1h ['$' for help]
$$ <
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000

CNCaddicted, 2021-04-21:
My 3018 Pro is using 6 limit switches and homes to the lower left corner. You may want to compare your settings to mine.
[CTRL+X] < Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
$X < [MSG:Caution: Unlocked]
ok
$$ < 
$0=10
$1=25
$2=5
$3=5
$4=0
$5=1
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=25.000
$25=500.000
$26=250
$27=3.000
$30=1000
$31=0
$32=0
$100=1600.000
$101=1600.000
$102=1600.000
$110=1000.000
$111=1000.000
$112=800.000
$120=10.000
$121=10.000
$122=10.000
$130=223.000
$131=140.000
$132=23.000
ok

SWIN2121, 2021-04-21:
Good morning I have my $100 $101 $102 set at 400.000  $110 $111 $112 set at 1500.000 no limit switch  I set the job to run  push send  now the spindle go off the table at the bottom  I use the lower left for home

From: CNCaddicted
Sent: Wednesday, April 21, 2021 5:25 AM
To: Denvi/Candle
Cc: SWIN2121; Author
Subject: Re: [Denvi/Candle] I Need Help (#476)

You may want to compare your settings with mine. I use homing/limit switches that home to the lower left hand corner.
[CTRL+X] < Grbl 1.1h ['$' for help]
$$ <
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000

CNCaddicted, 2021-04-21:
My attempt to delete the first message where $130-132 were 200 apparently failed. The second set of values is correct. Try this: put a piece of paper on the platform/wasteboard. Bring the tool down to within a couple of mm of the paper and make a mark. Make sure you are in metric mode (G21) not English (G20). [ that could be your entire problem]. Then zero the X & Y. Then in the Console Send box type in G0 x50.8. The tool should move 2 inches (50.8mm) to the right. Then enter G0 Y50.8. The tool should move up 2 inches. If not, you may want to change $100-102.

CNCaddicted, 2021-04-21:
If you want to work in English units you can set $13=1 and Candle will read out in inches rather than mm.

SWIN2121, 2021-04-21:
I put GOX50.8,    in the console box  error shows up   I set my home bottom left  set x&y  Use paper  press seed and the spindle goes to the bottom and i have to about the carve

SWIN2121, 2021-04-21:
Does anybody have the  Original $ codes i wood like to see them THANKS

CNCaddicted, 2021-04-23:
That is supposed to be G”zero” not G”oh”.

closed 2022-05-07 by Denvi