Denvi/Candle #481: Candle on Mac Big Sur 11.3.1

ahartman, 2021-05-16:
Candle 1 works fine on my Mac.
Candle 2 crashes at startup, see attached screen image.
[image]

Denvi, 2021-05-18:
This repo hasn't Mac builds.
What is "Candle 2"?

marsjo, 2021-05-19:
I run into the same problem.
You can get QT5 e.g. via brew: `brew install qt@5` and then copy the missing framework into the App Package. In my case: `cp /usr/local/Frameworks/QtNetwork.framework /Applications/Candle2.app/Contents/MacOS/Frameworks`

This fixed it for me. Let me know if you need some more help. I just saw that there is a branch related to macOS Builds and I hope we will get better support for macOS soon - would be really nice!

ahartman, 2021-05-20:
Dear Denvi,

An earlier message from vol4enok pointed to me to a Google drive location: https://drive.google.com/drive/folders/1AGgX804tFUevi9AKxamSViYKNyABAJ_W. There I found Candle_1.2.9.1b.dmg and Candle 2_2.23.dmg. It is that Candle2 that crashes.

Kind regards, ahartman, belgium

ahartman, 2021-05-20:
Dear marsjo,

Adding the QtNetwork.framework worked for me as well.

Denvi, 2021-05-20:
> Dear Denvi,
> 
> An earlier message from vol4enok pointed to me to a Google drive location: https://drive.google.com/drive/folders/1AGgX804tFUevi9AKxamSViYKNyABAJ_W. There I found Candle_1.2.9.1b.dmg and Candle 2_2.23.dmg. It is that Candle2 that crashes.
> 
> Kind regards, ahartman, belgium

Hi, ahartman.
Looks like Candle 2 is a fork, i can't help with it.

marsjo, 2021-05-20:
Hi Denvi, 

that's correct and I am not sure why it is there. There is also some branch doing comparison between Candle 1 and Candle 2. I used Candle 2 just because there was a ready compiled macOS App. Unfortunately I could not get it to work with my esp32-mpcnc.
Yesterday I successfully compiled your project using an existing PR and it works perfectly fine on macOS. 
Candle is by far the best CNC Software I could find and I am really happy that it works under macOS as well.

Hiworld123456, 2021-09-14:
> Hi Denvi,
>
> that's correct and I am not sure why it is there. There is also some branch doing comparison between Candle 1 and Candle 2. I used Candle 2 just because there was a ready compiled macOS [...]

Hello, I would like to ask you, why does my candle not display the path after putting the nc file? I have no idea to troubleshoot the problem.

closed 2025-09-09 by github-actions[bot]