Denvi/Candle #485: User functions not retained after restart in ver:1.2.13b_x64

dart1280, 2021-05-29:
**Versions**
ver:1.2.13b_x64
1.1f

**PC info**
win10 64bit
- OpenGL version:no idea

**Describe the bug**
after altering the user command buttons all edits are lost after exiting the program. I tried pressing 'set to defaults' as well...same result, all functions return to original and my edits are lost.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'settings
2. Click on 'any field to edit
3. Scroll down to restart the software'
4. See error

**Expected behavior**
It should remember my edits, as it does in ver 1.1.7

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Example files**
Add G-code files that cause problems.

**Additional context**
Add any other context about the problem here.

Denvi, 2021-06-01:
I can't repeat the above behavior, everything works as intended.

dart1280, 2021-06-01:
Um!   So if you edit a user command it remembers your edit after
re-starting the program?   Mine doesn't ... 100% of the time.

Candle 1.2.13b
Win10 64bit

It's not a biggie, I'll go back to 1.1.7.

Thanks anyway.

closed 2025-09-09 by github-actions[bot]