Denvi/Candle #487: Machine and work position errors

ghrrum, 2021-06-05:
**Versions**
 - Candle version: 1.1.7 and 1.2.9b
 - Firmware version: 1.1h

**PC info**
- OS: Win10
- OpenGL version: 3.1.0

**Describe the bug**
Machine position and work position do not change with spindle position, all other features appear to work. Repoduced issue in 1.2.9b in addition to 1.1.7,. Have reinstalled drivers to check that's not the issue (it's not). Checked position using Putty to confirm the position shown is correct (it's not). When attempting to zero out the machine to the work it seems to choose other positions at random

**To Reproduce**
Uniform issue on my end to the point Candle is not useable.

ondras12345, 2021-06-05:
Hi, please show us your Grbl settigns (`$$`). I would focus on [`$10`](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#10---status-report-mask), I have it set to `$10=1` on my machine.

ghrrum, 2021-06-06:
ondras12345,
Nailed it, was set as $10=2 on mine. Changed to $10=1 and issue resolved.

Thanks very much for the help, I'm off to work on seeing if I can get a circuit board to cut now.

closed 2021-06-06 by ghrrum