Denvi/Candle #488: Version 1.2b on Raspberry Pi?

mcinquino, 2021-06-06:
Hello, 

Has anyone got version 1.2b running on a Raspberry Pi 4? I would like to use some of the new features in it.. primarily the shortcut keys. I tried to compile but got errors. I have version 1.1.8 running on the pi but that did not require the compiling process. Was wondering if there was a simpler way to get it running on a pi?

I followed the instructions shown from wiki here: https://github.com/Denvi/Candle/wiki/Compiling-for-Raspberry-Pi-4

ondras12345, 2021-06-07:
Hi,
I think building from source is currently the best way to go.
You're probably just missing a few dependencies. Please post the exact errors you got.

mcinquino, 2021-06-09:
Hello,

I made the changes to the 2 files as outlined. Below is what I get when I qmake and make. 

pi@raspberrypi:~/Downloads/candle 12b/Candle-1.2b/src $ qmake candle.pro
Project WARNING: GL ES detected. VAO will be disabled.

pi@raspberrypi:~/Downloads/candle 12b/Candle-1.2b/src $ make
g++ -c -pipe -Z7 -Fdrelease\candle.pdb -O2 -Wall -W -D_REENTRANT -fPIC -DUNIX -DGLES -DsNan="65536" -DAPP_VERSION=\"1.2b\" -D_USE_MATH_DEFINES -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtOpenGL -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o main.o main.cpp
g++: error: unrecognized command line option \u2018-Z7\u2019; did you mean \u2018-Z\u2019?
make: *** [Makefile:773: main.o] Error 1

Thanks,
Mike

mcinquino, 2021-06-10:
Thank you. I will give this a try.

ondras12345, 2021-06-11:
To make plugins work (hopefully):
```sh
ln -s ../scriptbindings/plugins/script src/candle/script
ln -s ../candleplugins src/candle/plugins
```

yeyeto2788, 2022-02-04:
Hey @mcinquino!

I installed it on an x64 machine using a [tool I created](https://github.com/yeyeto2788/cnc_setup_ansible) you can see the screenshot of it running. You may want to give it a try, I indeed used the `grbl_1_1` on my tests but I don't see why it shouldn't work.

closed 2025-09-09 by github-actions[bot]