Denvi/Candle #494: floating window, translation, probe ...

milan-007, 2021-06-13:
**Versions**
 - Candle version: 1.2.b yesterday (12.6.2021 ) build from source via qtcreator
 - Firmware version: china 3018 upgaded from source to 1.1.h

**PC info**
- OS: opensuse 15.2
- OpenGL version: ...

**Describe the bug**
For first, I thank you for useful work....
Whea I build Candle, I faced few warnings and one errors, where `override ` miíssings ...  

When I test new floating window, I cannot anything enter to floating Console window, because immediatelly main form grab a focus..

As target in `candle.pro ` is defined Candle, but translations are candle_xx_xxx ( see case of first character). maybe on Windows it is not error, but on linux is ... As solution I rename target to candle or rename all translations files in candle.pro

Probe .... on my machine heightmap does not work completely. When I send 
```G38.2Z-1 F100 < [PRB:0.000,0.000,-0.955:1]
ok
G0X0Y0F100 < ok
Z2 < ok
G38.2Z-1 F100 < [PRB:0.000,0.000,-0.928:1]
ok
Z2 < ALARM:4
ok
G38.2Z-1 < error:9
```
via console, it is beatifully seen. `Probe` command on heightmap send G38.2Z-1 -or Z2 after G38.2  ( as I view via console) without Fxxx command. I do not know where problem is, but after first connection tool-PCB give error:9. Maybe this I Can setup by machne or something in program .. but I do not know

Probe command send : 
```
G21G90F100G0Z1 < ok
G0X0Y0 < ok
G38.2Z-1 < ALARM:5
[PRB:0.000,0.000,0.000:0]
ok
G0Z1 < error:9
G0X0.000Y0.000 < error:9
G38.2Z-1 < error:9
G0Z1 < error:9
G0X1.000Y0.000 < error:9
G38.2Z-1 < error:9
G0Z1 < error:9
G0X1.000Y1.000 < error:9
G38.2Z-1 < error:9
```
**To Reproduce**
Steps to reproduce the behavior:
1. Go to create heughtmap
2. Click on Probe
3. tool goes down one time, then I 
4. See error 9 and machine locks

**Expected behavior**
Heightmap are completed

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Example files**
Add G-code files that cause problems.

**Additional context**
Add any other context about the problem here.

Denvi, 2021-06-13:
Hi, milan-007.

Before creating new issue:
Candle works with CNC controlled by GRBL firmware, many problems can be solved by using proper version of GRBL, using proper configuration.

Please read GRBL wiki:

GRBL v0.9-: https://github.com/grbl/grbl/wiki
GRBL v1.1: https://github.com/gnea/grbl/wiki

...

ID | Alarm Code Description
-- | --
4 | Probe fail. The probe is not in the expected initial state before starting probe cycle, where G38.2 and G38.3 is not triggered and G38.4 and G38.5 is triggered.
5 | Probe fail. Probe did not contact the workpiece within the programmed travel for G38.2 and G38.4.

Looks like something wrong with your probe connection.

milan-007, 2021-06-14:
Thank you for reply, I look for alarm in detail ...

milan-007, 2021-06-14:
you are true - the probe was a little high ... now it works fine.. thank you.

closed 2022-05-07 by Denvi