Denvi/Candle #495: Z Probe issue

cnctom, 2021-06-14:
I'm new to this group so I'm sorry if I am approaching this in the wrong way but I am looking for help ideas.  I am not new to CNC, i Have a large 4' x 4' BobCnC for wood and I use Cut2D design software and UGC to run the Gcode.  No problem.  I recently purchased a SainSmart 3018 Prover which uses Candle and I have loaded Candle ver. 1.1.7.  Everything seems find except the Z prober.  I have put in the Gcode modifications outlined and I am within 1mm of the origin but when I initiate the Z probe I get no Z axis movement and the console puts out some gcode then "MSG: Reset to Continue"  Since the Z prober is not working I can not create the HeightMap which is the #1 reason I purchased this as I plan on only making PCB's with it.
I ended up loading the beta version of Candle 1.2.13b_x64 and got very different but still fail results.  When I initiate the probe the Z axis moves down but does not detect the prob and does not stop moving down.  I had to hit the emergency stop button.

When I re-downloaded the 1.1.7 code it went back to the initial results.

I have checked the continuity of the probe puck and leads, its all good.  No shorts and no opens.  I'm at a loss now on the issue, unless I just have a bad controller.

Thanks

Tom

ondras12345, 2021-06-15:
Hi,
what version of Grbl do you run on the machine? Please post your [Grbl settings `$$`](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration).
What Z-probe commands do you use? You can check this in your Candle settings.
For now, I would recommend running the [`G38.2`](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g38) manually with a slower feed rate, above the machine bed. You can try to short circuit the probe wires manually to see what happens.
How did you wire up the probe to the controller board?

milan-007, 2021-06-16:
probe is optional ... have you something on wires from control unit (crocodiles etc..) ?

cnctom, 2021-06-16:
I am told the unit shipped with grbl 1.1, but I don't know how to check that.  The unit is disassembled at the moment as I was rechecking the connections and inspecting the control board.  Everything looked good.  I will re-assemble later today and fetch the $$ output as well as try your G38.2 suggestion.  The probe command settings are currently G21G91G38.2Z-30F100; G0Z1; G38.2Z-2F10 per a setup and calibration video.  This unit comes with a probe and has a specific port to attach it to so I don't need the typical solder to A5.  The probe comes with the unit and has a puck and a clip.  As I said in my original post I have checked for both shorts and continuity and all is good.  I have also intentionally shorted the probe during the probe operation and nothing happens.  More to follow....

ondras12345, 2021-06-16:
> I am told the unit shipped with grbl 1.1, but I don't know how to check that.

Grbl version is printed to the Console window when the machine starts.

> as well as try your G38.2 suggestion

Don't forget that you need to specify the whole command, not just `G38.2`. I would try something like this (not tested):
```
(switch to incremental mode)
G91
(Try to probe 10 mm down with a feed rate of 20 mm per minute.)
(This should give you enough time to play with the probe wires.)
(The machine should stop moving once you short them together.)
G38.2 Z-10 F20
(go back to absolute distance mode)
G90
```

> I have also intentionally shorted the probe during the probe operation and nothing happens.

This is strange. If the probe is wired correctly, the machine should stop moving. Maybe take a look at [`$6`](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#6----probe-pin-invert-boolean), but I don't think it would behave like this if the probe signal was inverted.

cnctom, 2021-06-18:
Grbl 1.1f

 

$$ < $0=10

$1=25

$2=0

$3=2

$4=0

$5=0

$6=0

$10=3

$11=0.010

$12=0.002

$13=0

$20=1

$21=1

$22=1

$23=0

$24=25.000

$25=500.000

$26=250

$27=1.000

$30=10000

$31=0

$32=0

$100=793.650

$101=796.020

$102=798.400

$110=2000.000

$111=2000.000

$112=600.000

$120=10.000

$121=10.000

$122=10.000

$130=260.000

$131=158.000

$132=34.000

Ok

 

I am getting some success with G91 vs. G90.  The probe kinda works with G91, but not with G90.  Yet it seems to be very finiky about where the tool is to work and I can’t figure out the specifics.  I find it must be at the origin, which is fine, but it seems picky about the Z hight and I’m not sure what the deal is.  I’m wondering if I have a soft limit issue or something.  It’s definitely not on the hard limit switches.

 

Thx

 

Tom

ondras12345, 2021-06-18:
> I’m wondering if I have a soft limit issue

You can disable soft limits [`$20=0`](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean). But running into a soft limit should cause the machine to go to an alarm state.

cnctom, 2021-06-18:
I too am beginning to wonder if I’m hitting a soft limit.  I will try the setting change.

Tom Riggins
Sent from my iPhone, pardon the fat fingers

cnctom, 2021-06-23:
Just reporting back in.   I have achieved success.  I think their might have been a couple problems, but the key was disabling “soft limits”.    The G90 code would not work but I have not tried it with the soft limits off.  This YouTube tutorial guided me to turn on and set Soft Limits:  https://www.youtube.com/watch?v=u4Mcjz7IP10 <https://www.youtube.com/watch?v=u4Mcjz7IP10&t=249s> &t=249s at time point 5:24. Clearly not correct or at least not clear.  I will have to play with the Z limits to see if they were just too restrictive.

 

I ended up using the following code for the prober.

 

G91G21G38.2Z-20 F100

G92 Z00.00

G0Z2

M02

 

At any rate I achieved a successful PCB isolation Sunday evening.  I still have to work on other tuning with speeds and feeds to get a cleaner board, but the “Height Map” was imperative.

 

My other CnC has Closed Loop Servos vs. just steppers, I know understand the value of the closed loop servos.  Maybe I can figure out an upgrade later.

 

 

More tuning to go but I’m functional now.

 

 

Tom

closed 2022-05-07 by Denvi