Denvi/Candle #498: FR: allow plugins to generate gcode programs and/or check if queue is empty

positron96, 2021-06-20:
**Is your feature request related to a problem? Please describe.**
I am trying to make a plugin that generates some Gcode programs. My first attempt is to produce a parametrized array of drill holes. 
It seems I cannot do this the right way. The only way to do this is to spam `app.sendCommands` calls that will queue codes into sending queue. There is undesirable because little control is given to plugin once all these calls are made. It would be desirable to e.g. cancel the remaining gcodes if there is an error in the middle of the run. Also, spamming hundreds of `sendCommands` at once just does not seem to be the correct way to do it.

**Describe the solution you'd like**
Add to `ScriptFunctions`:
1. Expose API for plugins to generate gcodes into program area (where opened files usually go). This would fit nicely into current workflow: plugin generates a program, user has control to start/pause/abort it, apply heightmap correction if needed, check validity with grbl, etc. As far as I understand, right now access to `m_currentModel` data is private in `frmmain`.
2. Expose API for plugins to check current size of send queue (and sent queue). This way plugin would be able to send new gcode when previous are executed and not flood queue with the hundreds of gcodes. Again, access to `m_queue` and `m_commands` is private.

**Describe alternatives you've considered**
Currently I'm just calling app.sendCommands when button of plugin is pressed.

**Additional context**
Btw, adding scripting to candle is totally awesome!

positron96, 2021-06-20:
In the same spirit, exposing jogging functions to plugins would allow custom controllers support, e.g. gamepads (via QGamepad API) or cncjs web-pendants compatibility.

closed 2025-09-09 by github-actions[bot]

Denvi, 2025-10-08:
Exposed following script functions/properties:
- app.newFile()
- app.loadFile(...)
- app.saveFile(...)
- app.device.commandsLength
- app.device.queueLength

Denvi, 2025-10-08:
Not necessarily, you can load array of commands directly.
Like:
```js
app.loadFile([
    "G21G90",
    "G0X0Y0Z0",
    "G1X10F100",
    "G1Y10F200",
    "G0Z1",
    "G0X0Y0",
    "M2"
]);
```

positron96, 2025-10-08:
Oh, this is great, thank you!