Denvi/Candle #502: Hints of User Commands

SupayM, 2021-07-20:
**Versions**
 - Candle version: 1.2.14b

**PC info**
- OS: Windows 10

**Describe the bug**
The hint of user commands is showed in only one language, not changed with user interface.
Maybe just a mistake of mine.

Tryed to change the language.

If possible, coudl be nice to personaliza the hint on the user commands.

Or if I'm doing something wrong, please let me know.

Thanks

BTW, I'm new in this journey.

Denvi, 2021-07-21:
Hi, there is no user commands hints translate option for now. 
You can change any hint any time you want in settings window.

SupayM, 2021-07-21:
It could be a nice add then.

also, wat i need to do, to help on translations from English to Spanish, and to English?

i'm new at github, but I love to help this awesome project

Thanks

closed 2025-09-09 by github-actions[bot]