Denvi/Candle #505: Heightmap, Cant get it to work.. Please help me

geremyel, 2021-08-10:
**Versions**
 - Candle version:1.1.7
 - Firmware version: 1.1h

**PC info**
Windows 10

**Describe the bug**
A clear and concise description of what the bug is.
The heightmapping will not work. The z probe works fine, when I tried the height map, it would touch the board, then move up about 5mm, then alarm and error code 9. I don't know what to do. 
I've been working on this issue for the past 8 hours, since none of my pcbs are coming out properly due to uneven surfaces, and I just accidentally touched the ground of the probe on the ground of the main motor and now my cnc is totally dead. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Heightmap, set up
2. Click on Probe
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
I just want it it work. If anyone can help me figure out what I need to replace now also, that would be great..
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Example files**
Add G-code files that cause problems.

Denvi, 2021-08-10:
Hi, post screenshots please: heightmap settings, app window before & after "Probe" button click.

geremyel, 2021-08-10:
I reflashed grbl onto the board I have via USBtinyISP, and the main functions are working, but I cannot get the z probe not the heightmap to work. 

Here is the error when I try to use the z probe:
S9000 < ok
G91G21G38.2Z-50F100 < ALARM:4
ok
G92 Z14.09 < error:9
G0Z5M30 < error:9

Heightmap: 
-29.66;-41.62;60.93;87.07
3;2;-1.00;1.00
0;4;4
-65;-65;-65
-65;-65;-65

geremyel, 2021-08-10:
Screenshots before and after z probe
https://ibb.co/875cpQz
https://ibb.co/LnJxY1C

geremyel, 2021-08-10:
Before and after heightmap: 
https://ibb.co/r3r1JqT
https://ibb.co/RTMVgJM

Denvi, 2021-08-10:
Looks like your probe pin is always triggered, try to type $6 = 1 or $6 = 0 command in console. Check probe connections.

geremyel, 2021-08-10:
> Looks like your probe pin is always triggered, try to type $6 = 1 or $6 = 0 command in console. Check probe connections.

I tried changing that value between each 1/0, but no luck. 
My probe circuit is extremely simple, with duponts -> alligator clips onto board / spindle. I'm pretty sure there a pullup resistor on the board I have, which is why I didn't incorporate that. 

At first, the z probe was working fine, but the heightmap wasn't. It seems that after the short happened, the z probe no longer works. My thought is that there may be either physical hardware damage, or software corruption. I reflashed grbl onto the board, and still no luck. I don't see any damage, and as far as Im aware, the z probe goes to pin A5 of the Atmega328p. I've never really ran into an issue where a single pins functionality was rendered unusable. Its always been the whole board, or a physical component. 

Is there anything else on the computer software side to try? 

Ive tried changing invert probe pin values, Ive tried enabling/disabling soft limits in ugs, I've tried switching the probe leads. I checked continuity, there definitely is between both the board and pin and spindle and pin. 

Thanks

geremyel, 2021-08-11:
New GRBL board fixed z probe issues.. I think this may have been a hardware/atmega firmware issue. Thank you for the help

geremyel, 2021-08-11:
Update: Z probe works, but heightmap still doesn't. The bit is driven straight into the board and keeps going until it breaks/I stop it. Using same heightmap from before

Denvi, 2021-08-11:
Do you use "Zero Z" command before heightmap probe?
Check "Show G-code program commands", "Show Ui commands" under "Console" group in Settings window and post screenshots before/after heightmap probe please.

geremyel, 2021-08-11:
> Do you use "Zero Z" command before heightmap probe?
I did previously, but I forgot to do that with the new board I got, I did the z probe then got excited and overlooked the zero z. It works now. 
I really appreciate the help. 

One last thing, not sure if I should start a new topic or not, but the spindle is not starting when I press send with this new board. It does start if I manually press the spindle button, but it seems that its on max rpm, and changing the rpm slider doesn't seem to do much though I don't have a tachometer or anything to reliably measure it. I'm hoping that there's maybe a simple command Im looking over or something.

geremyel, 2021-08-11:
Reset seemed to fix it. Thank you!

geremyel, 2021-08-11:
For anyone that is having the same trouble:
1. Ensure proper z probe circuit/connections
2. Ensure that your z-probe works
3. Ensure that you "Zero Z" axis prior to starting height map, every time.

The problem must have been a fault in either the hardware, or some corruption in the atmega328p on the "Annoy Tools" board, because a replacement board from Amazon works fine now. If you follow these steps, and also try inverting the probe high/low with $6=1/0, then it wouldn't hurt to order a replacement board for ~$25 from amazon (or making one yourself if you have the stepper drivers), and testing it out with that. If anyone has the same problem with the "Annoy Tools" board, I'm curious to know if there is some design flaw in them, or if I did something to mess it up.

closed 2021-08-11 by geremyel