Denvi/Candle #507: Candle "jog" buttons deactivated at hold for changing to finishing cutting bit.  Cannot raise Spindle

Rroe34293, 2021-08-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Rroe34293, 2021-08-15:
Is there a way to activate the Candle Software buttons when my program reaching the halfway point and has paused to begin itself on a Finish cut?     I need to change cutting bits at this point and cannot raise the spindle because the Candle Jog buttons are not active.    Is there a way to activate these to allow raising the spindle for clearance to change out a cutting bit?

itsMe2021, 2021-09-27:
Cannot get any version of Candle jog buttons to work! Used my usual laptop with GrblControl running…and a new laptop with only Win 10 included apps.
Again Candle 2.3 and 2.2…the jog buttons do not function.
Cannot be com issues, new laptop, tried 3 usb ports..buttons do not function!

ondras12345, 2021-09-27:
> Candle 2.3 and 2.2

There is no such version. See https://github.com/Denvi/Candle/releases
This is probably because you are using Grbl firmware older than v1.1. [README](https://github.com/Denvi/Candle#downloads) says:
> (for GRBL v0.9 and below use Candle 1.0)

Denvi, 2022-05-07:
#521

closed 2022-05-07 by Denvi