Denvi/Candle #508: What are the build instructions for the Experimental branch?

martinhansdk, 2021-08-16:
**Versions**
 - Candle version: Experimental]

**PC info**
- OS: Linux

**Description**
I'd like to contribute to the development. It seems like the development is currently happening in the Experimental branch which has diverged considerably from the master branch. I have figured out how to build the master branch, but the Experimental branch has me stumped. I have downloaded Qt 5.4.2, but my Kubuntu system already has a newer version of Qt installed (inevitably, it is based on Qt), so qmake seems to always want to use the system version instead of 5.4.2. I have tried various thing with qtchooser and the QT_SELECT environment variable, but qmake always uses a mix of 5.4.2 and the system version:

```
$  QT_SELECT=5.4.2 ~/Qt5.4.2/5.4/gcc_64/bin/qmake candle.pro
Updating 'camera/translation_ru.qm'...
Removing translations equal to source text in 'camera/translation_ru.qm'...
    Generated 9 translation(s) (9 finished and 0 unfinished)
Updating 'usercommands/translation_ru.qm'...
Removing translations equal to source text in 'usercommands/translation_ru.qm'...
    Generated 10 translation(s) (10 finished and 0 unfinished)
Updating 'coordinatesystem/translation_ru.qm'...
Removing translations equal to source text in 'coordinatesystem/translation_ru.qm'...
    Generated 6 translation(s) (6 finished and 0 unfinished)
 martin  mahn-laptop  ~  extcode  Candle   Experimental  46+  1⚑  $  make
cd src/designerplugins/ && ( test -e Makefile || /home/martin/Qt5.4.2/5.4/gcc_64/bin/qmake /home/martin/extcode/Candle/src/designerplugins/designerplugins.pro -o Makefile ) && make -f Makefile 
make[1]: Entering directory '/home/martin/extcode/Candle/src/designerplugins'
cd cameraplugin/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /home/martin/extcode/Candle/src/designerplugins/cameraplugin/cameraplugin.pro ) && make -f Makefile 
make[2]: Entering directory '/home/martin/extcode/Candle/src/designerplugins/cameraplugin'
make[2]: Nothing to be done for 'first'.
make[2]: Leaving directory '/home/martin/extcode/Candle/src/designerplugins/cameraplugin'
cd customwidgetsplugin/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /home/martin/extcode/Candle/src/designerplugins/customwidgetsplugin/customwidgetsplugin.pro ) && make -f Makefile 
make[2]: Entering directory '/home/martin/extcode/Candle/src/designerplugins/customwidgetsplugin'
make[2]: Nothing to be done for 'first'.
make[2]: Leaving directory '/home/martin/extcode/Candle/src/designerplugins/customwidgetsplugin'
make[1]: Leaving directory '/home/martin/extcode/Candle/src/designerplugins'
cd src/candle/ && ( test -e Makefile || /home/martin/Qt5.4.2/5.4/gcc_64/bin/qmake /home/martin/extcode/Candle/src/candle/candle.pro -o Makefile ) && make -f Makefile 
make[1]: Entering directory '/home/martin/extcode/Candle/src/candle'
g++ -c -pipe -O2 -Wall -Wextra -D_REENTRANT -fPIC -DUNIX -DsNan="65536" -D_USE_MATH_DEFINES -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_UITOOLS_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_SCRIPT_LIB -DQT_CORE_LIB -I. -I../designerplugins/customwidgetsplugin -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtOpenGL -I/usr/include/x86_64-linux-gnu/qt5/QtUiTools -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtSerialPort -I/usr/include/x86_64-linux-gnu/qt5/QtScript -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o frmmain.o frmmain.cpp
In file included from /usr/include/x86_64-linux-gnu/qt5/QtGui/qtguiglobal.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/qtwidgetsglobal.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/qfiledialog.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/QFileDialog:1,
                 from frmmain.cpp:4:
frmmain.cpp: In constructor ‘frmMain::frmMain(QWidget*)’:
```
How do you build it?

ondras12345, 2021-08-16:
Take a look at this: https://github.com/Denvi/Candle/issues/488#issuecomment-858985048
I can't see an actual error message in your post, I think it was right below the text you copied.

martinhansdk, 2021-08-18:
Thank you, @ondras12345 . Using the instructions in #488 I have arrived at this:

On Ubuntu, download QT 5.4.2 from https://download.qt.io/new_archive/qt/5.4/5.4.2/, install it in `~/Qt5.4.2`.

```
cd Candle
qtchooser -install 5.4.2 /home/martin/Qt5.4.2/5.4/gcc_64/bin/qmake
export PATH=~/Qt5.4.2/5.4/gcc_64/bin/:$PATH
export QT_SELECT=5.4.2

# apply fixes from #488 
sed 's:-L../designerplugins/customwidgetsplugin/release -lcustomwidgets:-L../designerplugins/customwidgetsplugin -lcustomwidgets:' -i src/candle/candle.pro
sed 's:../designerplugins/customwidgetsplugin/release -lcustomwidgets:../designerplugins/customwidgetsplugin -lcustomwidgets:' -i src/scriptbindings/qtbindings/qtscript_custom/qtscript_custom.pro
sed 's:\(Q_DECLARE_METATYPE(QList<QSslError >)\)://\1:' -i src/scriptbindings/generated_cpp/com_trolltech_qt_network/qtscriptshell_QNetworkReply.cpp
sed 's:\(Q_DECLARE_METATYPE(QSslConfiguration)\)://\1:' -i src/scriptbindings/generated_cpp/com_trolltech_qt_network/qtscriptshell_QNetworkReply.cpp
    
qmake -r candle.pro
make
```

This fails with the following compile error:

```
camerawidget.cpp: In member function ‘void CameraWidget::setCamera(const QCameraInfo&)’:
camerawidget.cpp:88:5: error: ‘QCameraViewfinderSettings’ was not declared in this scope; did you mean ‘QCameraViewfinderPrivate’?
   88 |     QCameraViewfinderSettings fs = m_camera->viewfinderSettings();
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
      |     QCameraViewfinderPrivate
camerawidget.cpp:89:5: error: ‘fs’ was not declared in this scope; did you mean ‘ws’?
   89 |     fs.setResolution(m_resolution);
      |     ^~
      |     ws
camerawidget.cpp:90:15: error: ‘class QCamera’ has no member named ‘setViewfinderSettings’
   90 |     m_camera->setViewfinderSettings(fs);
      |               ^~~~~~~~~~~~~~~~~~~~~
make[2]: *** [Makefile:1023: camerawidget.o] Error 1
```

https://doc.qt.io/qt-5/qcameraviewfindersettings.html says that `QCameraViewfinderSettings` was added in Qt 5.5, so it looks like the README is wrong when it states that Qt 5.4.2 is needed.

What is the correct Qt version to use?

ondras12345, 2021-08-18:
Yeah, that section of README was last updated 4 years ago.
I used whatever the default version of Qt was on the system (probably 5.11.3 as it was a Debian Buster based OS).

Nandox7, 2021-08-18:
Likely is missing libs, all docs are quite old now.
For reference I got it to build to OSX Big Sur using latest LTS QT 5.12.11.

martinhansdk, 2021-08-20:
With a newer version of Qt this works. @Denvi would you accept a pull request that updates the build instructions and fixes the build scripts? 

I'd also like to add a github action that compiles on all three platforms to make sure that this keeps working. Would you accept a PR for that?

Another question: I get about 2000 depreciation warnings from the generated_cpp folder. Which tool generated the code in there? I'm wondering if I could make that tool generate the code automatically as part of the build process so that these files are always up to date with the version of Qt that is being used.

Denvi, 2021-08-28:
Hi, ok.
As for 'generated_cpp' it's https://github.com/phbaer/qtscriptgenerator

closed 2025-09-08 by github-actions[bot]