Denvi/Candle #510: visualizer freeze and the machinbe

ronisegoly, 2021-09-01:
**Versions**
1.2.9.1b
Windows
I am not sure about the exact details
When I use my CNC machine (Sainsmart 3018 and now 4030)
If I use an harder wood while milling the visualizer will stop updating while the machine still mills , and few minutes later the machine will stop progressing
If I reduce the step down or the spindle speed it will be OK.
My PC is very basic, could it be that when facing "harder work" it require more memory? if so then I can add memory, but want to make sure that's the issue

ondras12345, 2021-09-01:
The symptoms you describe indicate that your CNC machine loses connection with your computer. The visualizer stops updating because it does not get information about the machine's position, and the machine finishes all commands it has in it's buffer and stops working (or rather waits for another command).
I guess this could happen because your stepper motors have to work harder, cause more noise (electromagnetic interference) and thus interfere with the machine's ability to communicate with the computer.
Maybe try a different USB cable (preferably shielded).

> My PC is very basic, could it be that when facing "harder work" it require more memory?

I don't think this is memory-related. What you describe isn't "harder work" for the computer, the amount of commands it has to send is probably lower, and so is the number of movements / curves the visualizer has to show.
Can you still rotate the visualization or is it completely frozen?
What does the Windows Task Manager say about your memory usage?

ronisegoly, 2021-09-01:
Thanks for the detailed answer
The PC did not show memory issues, not even on Task manager which shows status
Do Candle have some log file?

ondras12345, 2021-09-01:
> Do Candle have some log file?

I don't think so. I use GNU/Linux, so this might be different on Windows, but when I start Candle in a terminal, it prints some limited debugging information to it's stdout. Also check the Console window in Candle's GUI.

ronisegoly, 2021-09-01:
I am monitoring now, with half step down 
If it halts I will see if buffer is 0
Meanwhile Windows memory looks OK

ronisegoly, 2021-09-01:
OK, it happened again, this time the spindle just stopped progressing, but still spinning
The visualizer still response ( i.e. can change view)
The buffer is not empty (see attached) and there is no indication on progressing status
https://photos.app.goo.gl/WgGsVUqWxpuYaaDMA
https://photos.app.goo.gl/wJCi2eJQ5TJTqBQX6
puzzled

ondras12345, 2021-09-01:
Don't expect the buffer information to update if the machine stops communicating. The machine's firmware would need to tell you what the state of the buffer is (or what commands it has already processed). To be honest, I'm not even sure what information the `Buffer: a / b / c` indicator shows is and how it is obtained. I don't think it's too important.

How about Status (under work coordinates) and Console?

ronisegoly, 2021-09-01:
The status was run
I wonder if I should try another SW

ondras12345, 2021-09-01:
You can certainly try another software, but I don't think it'll help. [bCNC](https://github.com/vlachoudis/bCNC) looks good, although I haven't tried it yet.
Next time this happens, check the console. It might contain some useful information and/or error messages.

ronisegoly, 2021-09-01:
I will try again tomorrow

ronisegoly, 2021-09-09:
BTW if anyone else encountered similar issue
It relates to Windows USB power management
Here is an advice I got.
"This is a power-saving feature that all windows computers have enabled by default and frequently causes this issue on longer projects. Here is a guide on how to make sure it is disabled."
https://www.pugetsystems.com/labs/support-software/How-to-Disable-USB-Selective-Suspend-1190/

albmef, 2023-06-12:
hi !
same issue here with my laptop and a 3018 toy.
the display stops updating but the gcode still runs in the background.
the serial com seems not robust at all. but since the last commit is 4 years old, i doubt that any improvement will be made.
this repo is dead imho.
bob