Denvi/Candle #511: 3018 CNC Not running

crwhite57, 2021-09-06:
Ok, I am new to this and I have read the post about a similar problem. I used EasyEDA to create my Gerber files, I then used FlatCam to create the alignment dill holes for the PCB I am trying to CNC. I load the code into Candle, move my spindle to x, y, and z-axis and zero it out. When I click on the send, the spindle starts and then stops. Candle shows the first 8 lines in the code as processed and ok, but the next 5 lines show sent and nothing is happening. Just so you know, I am running this on Windows 10 with lots of HD space, a 4ghz CPU, and 16Gbs of memory.

Please, can anyone help me? I need these PCBs for a project I am working on for my YouTube channel.

ondras12345, 2021-09-06:
Does your gcode program work in Check mode? Could you post the first 20 or so lines of the program and show us exactly where it stops?

crwhite57, 2021-09-06:
I just tried it with another gcode for the top layer and it did the same thing, 8 lines then it just shows sent for the next 5 and does nothing. These are the first 15 lines.

Command      State
1. G21              OK
2. G90              OK
3. G94              OK
4. F3.00              OK
5. G00 Z0.2000              OK
6. M03              OK
7. G4 P1              OK
8. G00 X21.8247Y54.9223              OK
9. G01 Z-0.0050              Sent
10. G01 X21.8204Y55.0031              Sent
11. G01 X21.8067Y55.0953              Sent
12. G01 X21.7841Y55.1858              Sent
13. G01 X21.7527Y55.2735              Sent
14. G01 X21.7128Y55.3578              Sent
15. G01 X21.6649Y55.4378              In Queue

That is where it stops and I get "[MSG:Reset to continue]" in the console. I have tried doing a rest and it just keeps doing the same thing.

ondras12345, 2021-09-06:
Above the `[MSG:Reset to continue]` line in the Console, there should be an explanation of what happened. Look for `error:n` or `alarm:n`.
Maybe Grbl does not like that you don't have a space between the X and Y coordinates.

crwhite57, 2021-09-06:
When I clear the Console, reset the code, and run it, all I get in the Console is that message.

crwhite57, 2021-09-06:
Here is everything in the Console by doing reset on the right and then unlock.
"[MSG:Reset to continue]
[CTRL+X] < ALARM:2
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
S9999 < error:9
$X < [MSG:Caution: Unlocked]
ok
S9999 < ok"

ondras12345, 2021-09-06:
This is probably caused by soft limits. Check [`$20`](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean) and [`$130`-`$132`](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#130-131-132--xyz-max-travel-mm).

crwhite57, 2021-09-06:
I think I found the problem. The gcode is telling it to put the bit at -0.05 and it doesn't like the -0.05. Do you have any idea how to fix this problem?

crwhite57, 2021-09-06:
I am closing this as I found the problem was with me the whole time. I didn't set the table height right and didn't prob the material height for the counter drill of the alignment holes. Thanks for your help.

closed 2021-09-06 by crwhite57