Denvi/Candle #513: General discussions questions, etc

Branch-Random, 2021-09-23:
I know this is more of a software development platform and isn't really a general discussion forum, but would it be appropriate to discuss Candle in general?

I don't have anything that I would tag as a bug or enhancement right now, but I do have some questions and I don't know where else to ask. Is there some place more appropriate to do that, or can we do it here?

And Denvi...  If the answer is "no", then I apologize, and please just delete.

Thanks,  Branch-Random

Branch-Random, 2021-09-28:
OK. So let me just get into a couple things.

First, I'm running the 1.2 experimental release, and I think it's fantastic. I was having translation language issues with earlier versions, and I gave 1.2b a try and never looked back. But that said, is Candle still an active project? I don't see a lot of traffic and I'm not sure if anyone is working on it. I'd be happy to help and contribute to the cause if I were qualified, but I'm really not. I've done lots of programming, but primarily at a lower machine code level.

So is there any insight into when 1.2 will be baked enough to be considered stable for release?

Second, all the documentation I could find is for 1.1.7 or earlier and I haven't found anything for 1.2 . I'm able to figure out most of what I need for 1.2 just by playing with it, but it would be nice to see how some of the features are really intended to be used.

Is there anything out there in the way of documentation for 1.2?

OresTruck, 2022-04-28:
I'd like to second this query.  I'm an amateur user, not a professional programmer or CNC'er , so I might be an inappropriate presence on Github, but I find that the questions and answers and general to-and-fro discussion on this site tend to be much more productive than other forums I've looked at.  
I built my own CNC machine as a Covid/Winter project last year.  It's GRBL/Arduino based and is only small (470mm x 350mm x 90mm working space) but seems to be pretty good for my woodworking projects.  Candle is my gcode sender of choice, and the 1.2 version is very attractive.  It has been  described as an experimental version, but it really looks the goods.  That said, I have some things that I hope contribute to the development discussions:

1 - When homing is in process, the Status indicator in the Device window stays at "Idle" until it flashes something (presumably "Homing") for an instant at the very end of the process before indicating "Idle" again.  This is harmless, but noticeable from an aesthetic point of view.

2 - The BIG RED Emergency Button is a great idea, but when I've used it the router just headed off in the positive Y direction until I pressed Reset.  I mention this in case anyone else has had the same problem, but I'd also mention that I've had intermittent problems with the positive-Y limit switch on my machine, so that might be a contributing issue. ( That said,  I re-emphasise my amateur status here.)  At the moment, the "Hold" and "Reset" buttons work so well I don't need the Emergency Stop button, so I don't display it. 

3 - The "Control" panel also presents an issue from an aesthetic/functional point of view.  I think of this as a job run-time panel, so some things seem out of place.  The Homing cycle, for instance, is a preparatory operation, not a run-time function, so I'd like to see it out of the Control panel.  That's probably true of the Check function as well.  I'm not familiar with the Sleep, Door or Mist functions, so I can't comment on them.  

4 - These quibbles aside, this comment is about supporting the query about documentation, and I have two related instances here:
- A previous post (#549) has asked about scripts in pop-ups  (as used in the Move Relative button in the User Commands  panel).  This is a really promising field of exploration for customisation.  Is it possible to document what's possible here for a non-stupid but unlearned amateur, or is this an area where a serious knowledge of a particular coding language is required?  Given that the aspirations of amateurs are not limited by realities, I'll understand if that's the case, but it would be nice to be advised of any reality-based limitations.
- In the settings for "Machine information" Candle 1.2 allows me to select Reference Position as (X- or X+),  (Y- or Y+) and (Z- or Z+).   I don't know the intention of these parameters.  The reference frame in the visualiser is shown as an X-Y rectangle whose size appears to be determined by the soft limit values given in the GRBL settings $130 and $131  (the Z component doesn't seem to be used here).  If I leave the Reference Position values as X-, Y- and Z-, which is the default, then the rectangle shows with its lower-left corner at the Machine origin.  In other words it is in positive X-Y space.  If I set the values as X+, Y+ and Z+ then the Reference rectangle is essentially a map of the X-Y space of my working area (i.e. my CNC table).  That seems useful, but I don't know if it's right or wrong, harmless or harmful, intended or contra-intentional.  That is where some user-aligned doco would be really helpful.

With all that said, I understand that Candle 1.2  is under development and that user doco must come last.  That means that asking for an instruction manual NOW is just jumping the gun, but I hope something like that eventuates because the project looks really promising.

Denvi, 2025-10-08:
Latest versions are stable and acquired new functions.
Documentation supplied with program now.