Denvi/Candle #514: Possible fix to GrblControl/Candle crash/freeze problem

pihnat, 2021-09-24:
We've been running GrblCntrl v0.9 and grbl v0.9j for a few years now and have constantly had random freeze/crash problems which I've investigated at various times. I feel that I may have found the cause and possible solutions. The info provided below should also be relevant to current versions of Candle/grbl so I hope you find it useful or at least a starting point to solve a freeze/crash problem you may have. Hopefully too this may address issues #510, #501, #461, #342, #338, #260 & #255. 

**Background:**
We run 16 student CNC machines (Sherline 5100) which are controlled by custom made stepper controllers. These used to be controlled by a DOS program with the controllers connected to the computers' parallel port using 25-pin printer cables. After a computer upgrade we went with GrblControl and Arduino Unos running grbl. The obvious solution was to install the Unos into the stepper controller boxes and connect them back to the computers using USB cables. Note that we don't have limit switches. The only signals between the Unos and stepper controllers are the X, Y, Z step and direction signals and the enable signal. The spindle motor is controlled manually on the front of the mills via a power switch and speed control dial.

**The problem:**
I found that most of the time there were no problems but occasionally at random times one would freeze. The visualizer screen would stop but the CNC machine would continue to run until the buffer in the Uno holding queued commands was emptied. Then it would stop with the spindle still running. GrblControl would totally lock up. The way to stop it was to go into Task Manager and end the task. However re-running GrblControl resulted in a serial port error and the only way to re-start things properly was to re-boot the computer.

**Possible solutions:**
So thinking it was an earth loop or something related to that I tried different solutions. Initially the Uno was powered by the USB cable so I tried powering it from inside the stepper controller instead. It didn't help. I tried opto-couplers in the step/direction/control lines between the Uno and the stepper controller board. It didn't work either. I couldn't make the system crash on cue, it just stayed a random problem. However I did notice when I was setting up the system that just an Uno by itself plugged into my desktop computer and running GrblControl was very stable.

Eventually by playing around with one of the CNC setups I finally found a way of reproducing the fault. It was related to the spindle motor. If I left the spindle motor off I could run a CNC machine without a workpiece all day and it would run flawlessly. However if the spindle motor was running and I re-ran the same G-code program over and over GrblControl would eventually crash. It would also crash if I turned the spindle speed to zero, switched it on and then turned the speed up to its operating speed. I also ran it as follows - with no workpiece & spindle motor off I would start the CNC running. Then during the run I'd switch the spindle motor on and off at various times and different speeds. Quite reliably during one of the switchings I could make GrblControl freeze. So the problem it seems is caused by electrical interference being generated by the spindle motor and being picked up by the USB cable/Uno arrangement. I also tried recompiling grbl setting the baud rate back to 9600 baud to make it less susceptible to noise but this also didn't work.

It looks like the USB cable picks up spikes from the spindle motor controller and the interfered data confuses either grbl running on the Uno or GrlControl/Candle running on the main computer. I haven't looked into whether it's the data being sent or data being received which is the cause. So some possible solutions:
**1.** if the problem is with the data being received by GrblControl then the serial port part of the software could be modified so if the received data doesn't make sense it should ignore it and request data again. I notice that in the current version of Candle there's an option in the settings called "Ignore error responses". Since I'm not running Candle I haven't tested if this solves the problem but this option might only apply to valid error codes being sent back from grbl not errors in the actual data packets being transferred.
**2.** if the problem is with the data being received by the Uno running grbl then a similar change to grbl software might help but this requires changes to the grbl source code.
**3.** another test I haven't tried but may work is to connect the Uno to the computer using **double shielded** USB cable. Issue #342 seems to imply that this might work. If anyone else tries it please report back indicating if it works or not.
**4.** the solution I've gone with in our case was to remove the Uno from the stepper controller box and mount it in a standard zippy box. I mounted a female 25-pin D-connector on the box and connected the Uno's I/O pins to it. Then I connected the box to the stepper controller using the 25-pin printer cable we used to use before. To connect the Uno to the computer I made up a very short USB cable about 120mm long. Being so short this appears to stop the spindle motor spikes from being picked up by the USB cable.

Once we're out of lockdown I'll keep testing this last solution but so far it seems to have worked.

ondras12345, 2021-09-24:
> if the problem is with the data being received by GrblControl then the serial port part of the software could be modified so if the received data doesn't make sense it should ignore it and request data again.

I think this issue is caused by electromagnetic interference on the USB bus, not on the serial port. This means that once it fails, Candle and/or Grbl cannot do anything about it. A driver on the PC is unable to communicate with the USB to UART converter (e.g. CH340). Candle is only reading from / writing to a virtual file (e.g. COM1) provided by the driver, and Grbl is using it's UART peripheral. Neither Candle nor Grbl can fix an issue with USB communication.

pihnat, 2021-09-25:
Yes, that makes a lot of sense since it's the packets of data being transferred on the USB cable which are being interfered with not the actual serial data. This would also explain why when I tried using optocouplers & changing the baud rate it didn't help either.

So that eliminates my possible solutions no 1 and no 2 just leaving nos 3 and 4.
Thanks

closed 2022-05-07 by Denvi

apachebcn, 2024-02-12:
> > Si el problema es con los datos recibidos por GrblControl, entonces la parte del puerto serie del software podría modificarse, de modo que si los datos recibidos no tienen sentido, debería ignorarlos y solicitar datos nuevamente.
> 
> Creo que este problema se debe a una interferencia electromagnética en el bus USB, no en el puerto serie. Esto significa que una vez que falla, Candle y/o Grbl no pueden hacer nada al respecto. Un controlador en la PC no puede comunicarse con el convertidor de USB a UART (por ejemplo, CH340). Candle solo lee/escribe en un archivo virtual (por ejemplo, COM1) proporcionado por el controlador, y Grbl está usando su periférico UART. Ni Candle ni Grbl pueden solucionar un problema con la comunicación USB.

I know the topic is old
I don't know if anyone managed to resolve the issue.
I suspect it is about interference on the cable or board.
My way to almost solve it casually is by using a USB hub
my cnc 3018 connected to my desktop computer stops after a few minutes
with the usb hub and connected to the same computer, it happens to me every so often doing entire jobs
I don't have any more cables so I can do the test with another cable.

> Yes, that makes a lot of sense since it's the packets of data being transferred on the USB cable which are being interfered with not the actual serial data. This would also explain why when I tried [...]

> I think this issue is caused by electromagnetic interference on the USB bus, not on the serial port. This means that once it fails, Candle and/or Grbl cannot do anything about it. A driver on the PC [...]

I know the topic is old
I don't know if anyone managed to resolve the issue.
I suspect it is about interference on the cable or board.
My way to almost solve it casually is by using a USB hub
my cnc 3018 connected to my desktop computer stops after a few minutes
with the usb hub and connected to the same computer, it happens to me every so often doing entire jobs
I don't have any more cables so I can do the test with another cable.

pihnat, 2024-02-12:
Rolling it up may not help but is worth a try. I don't think the magnet would help at all.
I did think of trying a double shielded USB cable but these are hard to find so I couldn't test it. Since we built the stepper controllers ourselves it was easy to remove the Arduino and put it in a separate box well away from the CNC machine.

The interference I found with our setup was the spindle motor on the Sherline CNCs. If we left the motor off and ran the whole job as a test it would never crash. As soon as we switched on the spindle motor it would crash, sometimes as soon as the motor was turned on, other times when it was part way through the job. It took quite a while to find that link.

Good luck with your tests.

apachebcn, 2024-02-20:
Hi People, I tell you my good news about this problem.

Today I discovered something very good and I don't think it's a coincidence.
In addition to the drill, the laser also gave me the same problem, the USB disconnection.
Today I used a USB extension cord, and this time without the USB hub which seemed to reduce those disconnection events a bit.
That extension cord is USB 3.0 with a cover. That lining looks like textile or synthetic or a mix of both, I'm not sure.
Cable is this: (sorry, it is not my intention to advertise this specific product nor do I have any relationship with it, just for your understand or investigation)
https://www.amazon.es/gp/product/B086ZGYCLD/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

It is quite likely that this cable has some metal shielding inside, because it is quite thick.
The fact is, I've been using the laser for more than 5 hours to make cuts in wood, and it hasn't stopped.
The normal thing was to suffer these constant disconnections, and when starting the connection of the machine with the computer's USB again, the "candle" application placed the cursor in the center, while the machine had the position in the last work order. (bad idea of the application, because if it maintained the position where the connection is broken, the problem would be much less)

I don't know if the drill will solve it and it's the magic of the cable.
It could be probable.
The CNC cable that connects with this extension is away from the controller, passing under the CNC structure.

In case anyone wants to make this same attempt.

In case I haven't explained myself well, the change consists of using the normal cable that I always use with the cnc, but having it separated and away from the cnc controller and the computer, and from this cable to the computer use the extension cable described.
It is connected to the blue USB 3.0 port, I don't know if that will be relevant.
But right now my cnc is working non-stop as if it never had any problems.

pihnat, 2024-02-20:
Great news.
I translated the web page you mentioned and it says the cable is "triple shielded" so yes, that should definitely stop interference being picked up. It confirms that the problem is definitely interference picked up on standard USB cables.

pihnat, 2024-04-11:
Hi all,
yes I can confirm that a very short USB cable also stops the crashing problem.
So the 2 solutions which work are
1. use a very short USB cable between the computer and Arduino and a longer 25-pin cable from the Arduino to your stepper motor controller or
2. if the Uno is close to or inside your stepper motor controller then use a triple shielded USB cable from the computer to the Arduino.

best regards

bruxoteles53-lang, 2025-10-02:
Olá colegas, sou brasileiro e tenho uma TTC 3018 pro. 
Minha máquina apresentou o mesmo problema. Sempre travando o candle durante a usinagem. 
Realmente o problema é a interferência eletromagnético do eixo Z, este, quando ativado, acaba interferindo na conexão USB. 

Solução encontrada:

Coloquei um eletroímã (anel de ferrite) no cabo USB, passando cerca de 4 voltas dentro do eletroímã e deixando o anel próximo a máquina. 
Acredito que conforme o comprimento do cabo serão mais ou menos voltas dentro do eletroímã. 
Problema resolvido. Minha CNC está trabalhando normalmente agora. 

Espero ter ajudado.