Denvi/Candle #517: Console command input field inactive

modsley, 2021-10-05:
**Versions**
 - Candle version: 1.1.7
 - Firmware version: 1.1f

**Describe the bug**
Port opened, jog control is working.
Input field for console commands is grayed, so I can't send console commands.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

**Additional context**
Add any other context about the problem here
[image]
.

ondras12345, 2021-10-05:
Try pressing `Scroll Lock`. Maybe you are in keyboard control mode.

modsley, 2021-10-05:
Yeah... When Keyboard control is checked console input is disabled. Thank you.

closed 2021-10-05 by modsley