Denvi/Candle #521: M0 4axis

DEST1981, 2021-11-20:
**Versions**
 - Candle version: [1.2.14b]
 - Firmware version: [e.g.: 1.3A]

**PC info**
- OS:Windows

Добрый день!
Хочу поблагодарить за отличный софт.

Прошивка GRBL ESP32 4 оси XYZA, плата моя, драйвера внешние.
Нашёл несколько багов/фич:
1. При использовании карты высот строится она не на оси Z а на хз какой, соответственно правки в G-код вносятся не верно. Для гравирования не подходит. В прошивке заблокировал ось А, карта строится нормально, функционирует тоже на отлично.
2. G-код M0 - пауза. Если находится в программе то вызывает глюки, станок зависает на удержании. В старой версии Candle 1.1.7 для замены инструмента я заменял код M06 на связку:
M5                             Останавливаем шпиндель
G0 Z30                       Поднимаем шпиндель
G0 X0 Y0                   Двигаемся в начало координат
M0                             Пауза (в этот момент меняю инструмент на другой и подключаю Z-щуп)
G38.2 Z-120 F100      Запускаю замер 
G10 L20 P1 Z0           Обнуляю координаты
M0                             Пауза (отключаю щуп)
G0 X0 Y0 Z5               Поднимаю инструмент от поверхности
M03 S12000              Включаю шпиндель
В новых версиях появилась возможность вписать код замены инструмента,но увы пауза все ломает. 
Происходит следующее:
Включается шпиндель. Срабатывает код M06, на экране появляется сообщение: Проведите смену инструмента нажмите паузу для продолжения. В это время отключается шпиндель, поднимается шпиндель, шпиндель едет в 0, запускается замер. Сообщение висит на экране. Если нажать ок на всплывающем окне программа предложит запустить кусок кода после М06. На кнопки пауза и прервать не реагирует. Если замер щупом проходит успешно программа зацикливается на Статусе Удержание.
**Versions**
 - Candle version: [1.2.14b]
 - Firmware version: [e.g.: 1.3A]

**PC info**
- OS:Windows

Добрый день!
Хочу поблагодарить за отличный софт.

Прошивка GRBL ESP32 4 оси XYZA, плата моя, драйвера внешние.
Нашёл несколько багов/фич:
1. При использовании карты высот строится она не на оси Z а на хз какой, соответственно правки в G-код вносятся не верно. Для гравирования не подходит. В прошивке заблокировал ось А, карта строится нормально, функционирует тоже на отлично.
2. G-код M0 - пауза. Если находится в программе то вызывает глюки, станок зависает на удержании. В старой версии Candle 1.1.7 для замены инструмента я заменял код M06 на связку:
M5                             Останавливаем шпиндель
G0 Z30                       Поднимаем шпиндель
G0 X0 Y0                   Двигаемся в начало координат
M0                             Пауза (в этот момент меняю инструмент на другой и подключаю Z-щуп)
G38.2 Z-120 F100      Запускаю замер 
G10 L20 P1 Z0           Обнуляю координаты
M0                             Пауза (отключаю щуп)
G0 X0 Y0 Z5               Поднимаю инструмент от поверхности
M03 S12000              Включаю шпиндель
В новых версиях появилась возможность вписать код замены инструмента,но увы пауза все ломает. 
Происходит следующее:
Включается шпиндель. Срабатывает код M06, на экране появляется сообщение: Проведите смену инструмента нажмите паузу для продолжения. В это время отключается шпиндель, поднимается шпиндель, шпиндель едет в 0, запускается замер. Сообщение висит на экране. Если нажать ок на всплывающем окне программа предложит запустить кусок кода после М06. На кнопки пауза и прервать не реагирует. Если замер щупом проходит успешно программа зацикливается на Статусе Удержание.

Denvi, 2021-11-20:
Приветствую.
Пока не могу повторить описанное Вами поведение.

Для анализа необходимы:
- часть G-code программы с командами смены инструмента;
- вывод в консоль с включенными флагами "Отображать команды G-code программы" и "Отображать команды интерфейса пользователя" в группе "Консоль" настроек программы.

Также для проверки прикладываю исполняемый файл программы с некоторыми исправлениями: [Candle_iss521.zip](https://github.com/Denvi/Candle/files/7574960/Candle_iss521.zip)
Файл для 64-разрядной ОС, если нужен для 32-разрядной, дайте знать.

XRay3D, 2021-11-20:
Мне кажется надо грбл отвязать от гуи и вынести в отдельный модуль - класс. Появится больше гибкости и проще вносить изменения в будущем. У меня уже есть некоторые наработки в этом плане.

DEST1981, 2021-11-20:
Во вложении весь код. Можете пробовать. [indicator_IN12 2pcs.txt](https://github.com/Denvi/Candle/files/7575366/indicator_IN12.2pcs.txt)
Во вложении весь код. Можете пробовать.
Сейчас ситуация такая:[image]
Либо виснет либо вылет.
Команды смены инструмента: M5
G0 Z30
G0 X0 Y0  
G38.2 Z-120 F100
G10 L20 P1 Z0
M0
G0 X0 Y0 Z5
M03 S12000

Denvi, 2021-11-21:
Спасибо, удалось повторить ошибку. 
Вот исправления: [Candle_iss521_1.zip](https://github.com/Denvi/Candle/files/7575870/Candle_iss521_1.zip)

Denvi, 2021-11-21:
> Мне кажется надо грбл отвязать от гуи и вынести в отдельный модуль - класс. Появится больше гибкости и проще вносить изменения в будущем. У меня уже есть некоторые наработки в этом плане.

Надо конечно, но пока есть более приоритетные задачи.

DEST1981, 2021-11-21:
К сожалению, работает все равно не корректно. 
Сейчас ситуация следующая: 
1.Программа запускается доходит до кода M06.
2. Всплывающее окно с сообщением про паузу и замену инструмента.
3. В фоне, в не зависимости от нажатого ок запускается код "Команды смены инструмента"
4. Пауза там не отрабатывает.
5. Запускается замер щупом.
6. Нажимаем ок.
7. Щуп находит высоту.
8. Дальше все, на кнопку пауза не реагирует, статус: "Удерживание (0)"
9. На сброс, разблокирование - реакция только в консоли. На кнопки движения, вкл шпиделя - реакции нет.
10. Понажимал на кнопку "Удерживание", разблокировался шпиндель.
В общем так как-то.

Denvi, 2021-11-21:
Может быть дело в прошивке конечно...
Для Grbl 1.1h и следующего списка команд смены инструмента:
```
M5
G0 Z30
G0 X0 Y0
M0
G38.2 Z-120 F100
G10 L20 P1 Z0
M0
G0 X0 Y0 Z5
M03 S12000
```
происходит следующее:
1) программа доходит до кода М06;
2) появляется окно с предупреждением про паузу, в фоне выполняются команды смены инструмента, на первой команде М0 производится остановка, жмем на кнопку "ОК" окна предупреждения;
3) меняем инструмент, подключаем щуп, жмем на кнопку "Удерживание", производится поиск нулевого положения по оси Z и сброс координаты, снова остановка;
4) отключаем щуп, жмем на кнопку "Удерживание", производится подъем инструмента и запуск шпинделя;
5) жмем на кнопку "Пауза", подтверждаем запуск оставшейся части программы, программа выполняется дальше.

> Либо виснет либо вылет.
Команды смены инструмента: M5
G0 Z30
G0 X0 Y0
G38.2 Z-120 F100
G10 L20 P1 Z0
M0
G0 X0 Y0 Z5
M03 S12000

Здесь потерялась одна команда М0.

DEST1981, 2021-11-21:
> Может быть дело в прошивке конечно... Для Grbl 1.1h и следующего списка команд смены инструмента:
> 
> ```
> M5
> G0 Z30
> G0 X0 Y0
> M0
> G38.2 Z-120 F100
> G10 L20 P1 Z0
> M0
> G0 X0 Y0 Z5
> M03 S12000
> ```
> 
> происходит следующее:
> 
>     1. программа доходит до кода М06;
> 
>     2. появляется окно с предупреждением про паузу, в фоне выполняются команды смены инструмента, на первой команде М0 производится остановка, жмем на кнопку "ОК" окна предупреждения;
> 
>     3. меняем инструмент, подключаем щуп, жмем на кнопку "Удерживание", производится поиск нулевого положения по оси Z и сброс координаты, снова остановка;
> 
>     4. отключаем щуп, жмем на кнопку "Удерживание", производится подъем инструмента и запуск шпинделя;
> 
>     5. жмем на кнопку "Пауза", подтверждаем запуск оставшейся части программы, программа выполняется дальше.
> 
> 
> > Либо виснет либо вылет.
> > Команды смены инструмента: M5
> > G0 Z30
> > G0 X0 Y0
> > G38.2 Z-120 F100
> > G10 L20 P1 Z0
> > M0
> > G0 X0 Y0 Z5
> > M03 S12000
> 
> Здесь потерялась одна команда М0.
> 
> Небольшое пояснение по поводу кнопок "Пауза" и "Удерживание":
> 
>     * "Пауза" отвечает за приостановку/продолжение отправки команд G-code программы, к команде M0 не имеет отношения;
> 
>     * "Удерживание" отвечает за оперативную остановку станка с использованием специальных команд "!", "~", определенных в прошивке GRBL, в состояние удерживания также можно перевести обычной командой M0.

closed 2021-11-21 by DEST1981

DEST1981, 2021-11-21:
Сейчас я пользуюсь версией 1.1.7, с этой версией проблем никаких, за исключением карты высот для 4х осевого станка.
Добавил недостающий код M0. Перепроверил работу в общем ситуация следующая: 
Появляется окно инструмент двигается в 0. Нажимаем ок, меняем инструмент и нажимаем "Удерживание", далее замер высоты, ещё раз "Удержание", а в конце нажимаем паузу.
Буду тестировать. 
А про карту высот что скажете?

DEST1981, 2021-11-21:
Итак:
Убрал галочку в настройках "Приостанавливать передачу файла по командам смены инструмента".
Запускаем код. 
Команды смены инструмента отрабатывают, М0 подтверждаю "Удерживанием" включается шпиндель и всё. Стоит на месте.
Поставил галочку в настройках "Приостанавливать передачу файла по командам смены инструмента". Всё тоже самое только выдаёт сообщение о том что нужно будет нажать паузу. По окончании выдаёт окно что будет выполнен следующий код: "" Кода нет. Висит на паузе и никуда не двигается. Нажатие на паузу, удержание, прерывание ничего не даёт. Вывести из тупика можно только нажав прерывание и потыкав в управление шпинделем.

Denvi, 2021-11-21:
> По окончании выдаёт окно что будет выполнен следующий код: "" Кода нет.

На первой смене инструмента кода не должно быть, это нормальное поведение (команды восстанавливают скорость шпинделя и позицию инструмента до прерывания в соответствии с программой, на момент первой смены инструмента ни скорость шпинделя, ни положение инструмента еще не были заданы).

[image]

Здесь не понятно, откуда взялась пара команд "M5" и "M2".

По карте высот еще не смотрел, устройств под 4 оси нет в наличии.
В целом, надо искать железку под вашу прошивку и тестировать, судя по всему, по разному они себя ведут.

DEST1981, 2021-11-22:
В общем смотрите, "M5" и "M2" появились когда я выводил программу из ступора.
Удалось таки запустить выполнение кода, но......
код смены инструмента такой:
M5
G0 Z30
G0 X0 Y0  
M0
G38.2 Z-120 F100
G10 L20 P1 Z0
M0
G0 X0 Y0 Z5
M03 S12000
В конечном итоге получается вот такой алгоритм:
Запускаем выполнение кода, ловим М6.
Нажимаем ок во всплывающем окне. Далее меняем инструмент нажимаем "Удержание" происходит замер Z.
Затем нажимаем пауза, во всплывающем окошке про команды нажимаем ок, Ничего не происходит. Нажимаем удержание. И О Чудо! Программа не в ступоре выполнение кода продолжается нормально. В другом порядке не работает. Попробовал ещё раз, в общем кнопка пауза должна быть нажата не последней. Т.е. можно сразу нажать. А потом тыкать в "Удержание" и всё норм.
К слову я вашей программой пользуюсь с конца 16го года, т.е. уже 5 лет. Первый контроллер был на ардуино. И все глюки которые я изредка ловил были связаны именно с паузой. Воспроизвести их мне не удавалось. Но при выполнении кода если в какой-то непонятный момент нажать паузу бывало что инструмент продолжал двигаться а программа стояла на месте, несколько фрез так закончили свою жизнь.

Denvi, 2021-11-23:
Пока едет ESP32, задам еще пару вопросов.

> Поставил галочку в настройках "Приостанавливать передачу файла по командам смены инструмента". Всё тоже самое только выдаёт сообщение о том что нужно будет нажать паузу. По окончании выдаёт окно что [...]

Сделайте, пожалуйста, снимок экрана окна программы целиком в момент когда будет выдано окно о том, что будет выполнен следующий код... и нажата кнопка "Ок", т.е. в момент, когда G-code программа должна продолжить выполняться.

[image]

К примеру такой снимок: 

[image]

> И все глюки которые я изредка ловил были связаны именно с паузой. Воспроизвести их мне не удавалось. Но при выполнении кода если в какой-то непонятный момент нажать паузу бывало что инструмент продолжал двигаться а программа стояла на месте, несколько фрез так закончили свою жизнь.

В версиях начиная с 1.2, такого глюка не должно быть, т.к. пауза функционирует иначе.

DEST1981, 2021-11-23:
Прогнал весь процесс.
В первый раз когда ловишь М06 - алгоритм именно такой как я указал.
В последующие разы нужно нажимать паузу последней. Иначе ступор.
Причём инструмент сначала возвращается к той точке в которой он поймал смену инструмента, а уж потом продолжает программу.
Всё это наблюдается с включённой галочкой приостанавливать передачу файла по команде смены инструмента.
Убираем галочку "приостанавливать передачу файла по команде смены инструмента" и пользуемся только "Удержанием" в таком раскладе,кроме инструмент сначала возвращается к той точке в которой он поймал смену инструмента, всё работает как нужно.

DEST1981, 2021-11-23:
[image]
Первый М06

DEST1981, 2021-11-23:
[image]
Второй раз.

DEST1981, 2021-11-23:
[CopperCAM.txt](https://github.com/Denvi/Candle/files/7588419/CopperCAM.txt)
G-код

Denvi, 2021-11-23:
> Всё это наблюдается с включённой галочкой приостанавливать передачу файла по команде смены инструмента.
> Убираем галочку "приостанавливать передачу файла по команде смены инструмента" и пользуемся [...]

Правильно ли понимаю:
1) С выключенным флагом "Приостанавливать передачу..." и первая и последующие команды смены инструмента работают без зависаний?
2) Опять же с выключенным флагом, команды возврата инструмента в точку команды M6, по большому счету, не нужны ( имеет смысл ввести соответствующий пункт в настройки, например "Возвращать инструмент в точку команды смены инструмента")?

DEST1981, 2021-11-23:
1. Да все верно. Если честно я вообще не вижу смысла во флаге "Приостанавливать передачу..." оператор сам должен решить что ему делать при смене инструмента.
2.  Я думаю имеет смысл ввести данный пункт "возврат инструмента в точку команды M6". Я для себя подобрал приемлемый способ ручной замены в нуле координат, т.к. если что то пойдет не так, я всегда смогу вернуться или начать сначала. Но например, если заготовка не плоская, при  перемещении можно запороть заготовку возвращаясь по короткому пути в точку команды М6.

Ещё я думаю есть смыл добавить всплывающее окно типа "Обнаружена команда смены инструмента. Выполнить код замены?" Не всегда требуется проводить замену. Ну например я рассверливаю отверстия кукурузой которой впоследствии провожу обрезку.
Есть смысл добавить чекбокс или флаг "игнорировать" на команды начала конца и смены инструмента, например прямо на против надписей. Это просто удобно, иногда одним инструментом работаешь а в программе разные куски со сменой на один и тот-же.

Ну раз пошла такая пьянка в новой версии поменялся смысл кнопок "Пауза" и "Прервать". Новую логику я понял, кнопки работают с передачей G-кода. Но..... есть довольно приличный буфер команд которые уже в контроллере, и это кнопки не прерывают работу сразу. Часть кода продолжает исполняться. Представляем себе ситуацию оператор замечает что что-то пошло не так. Если в старых версиях нажал "Прервать" или "Пауза" станок тут-же остановился, полностью или замер. То сейчас есть кнопка "Удержание", ну ок, паузу она заменила. А как прервать выполнение программы полностью. Кроме сброса ничего не поможет. А если нажать удержание а потом сброс, программа уходит в ступор.
Если честно я теперь не очень понимаю зачем вообще кнопки "пауза" и "прервать" - смысловая нагрузка на них нулевая. Я себе пока слабо представляю ситуацию в которой мне потребуется приостановить/прервать выполнение кода через неопределённое время. Если команды линейного перемещения то там уже половину стола изрежет в хлам пока остановится.

Denvi, 2021-11-24:
> 1. Да все верно. Если честно я вообще не вижу смысла во флаге "Приостанавливать передачу..." оператор сам должен решить что ему делать при смене инструмента.

Приостановка передачи позволяет произвести замену инструмента в полностью ручном режиме, по сути, в момент приостановки становится доступно управления станком без ограничений, актуально, к примеру, при ручной установке нуля по Z.

> 2. Я думаю имеет смысл ввести данный пункт "возврат инструмента в точку команды M6". Я для себя подобрал приемлемый способ ручной замены в нуле координат, т.к. если что то пойдет не так, я всегда [...]

Можно добавить подъем по Z на безопасную высоту, возврат к точке M6 идет в плоскости XY.

> Ещё я думаю есть смыл добавить всплывающее окно типа "Обнаружена команда смены инструмента. Выполнить код замены?" Не всегда требуется проводить замену. Ну например я рассверливаю отверстия кукурузой которой впоследствии провожу обрезку.
Есть смысл добавить чекбокс или флаг "игнорировать" на команды начала конца и смены инструмента, например прямо на против надписей. Это просто удобно, иногда одним инструментом работаешь а в программе разные куски со сменой на один и тот-же.

Добавлю, посмотрим, что выйдет.

> Но..... есть довольно приличный буфер команд которые уже в контроллере, и это кнопки не прерывают работу сразу. Часть кода продолжает исполняться. Представляем себе ситуацию оператор замечает что [...]
Если честно я теперь не очень понимаю зачем вообще кнопки "пауза" и "прервать" - смысловая нагрузка на них нулевая. Я себе пока слабо представляю ситуацию в которой мне потребуется приостановить/прервать выполнение кода через неопределённое время. Если команды линейного перемещения то там уже половину стола изрежет в хлам пока остановится.

Да, все так и есть. Для аварийной ситуации надо иметь большую красную кнопку где-нибудь рядом со станком. Команды для остановки в прошивке нет, так что или удержание/сброс или сброс (в 1.1.7 прерывание сделано через удержание/сброс с вагоном сопутствующих проблем и глюков).

DEST1981, 2021-11-25:
> Приостановка передачи позволяет произвести замену инструмента в полностью ручном режиме, по сути, в момент приостановки становится доступно управления станком без ограничений, актуально, к примеру, [...]
Я собственно в замене инструмента добавил код подъёма инструмента. Всё равно мне кажется это странное действие, возвращаться в ту точку откуда пришёл.
> Добавлю, посмотрим, что выйдет.
>
>
> Да, все так и есть. Для аварийной ситуации надо иметь большую красную кнопку где-нибудь рядом со станком. Команды для остановки в прошивке нет, так что или [...]
Вот ни разу не ловил глюки с кнопкой прерывание в версии 1.1.7 с паузой глюки были.
Не предусмотрел я большую красную кнопку кода плату разводил, да и портов свободных у меня не оставалось. Я рассчитывал на 4 оси с концевиками.
Опять-же у Вас есть камера, что означает что оператор может находиться не прямо рядом со станком. "Большая красная кнопка" должна быть и в программе.

Denvi, 2021-11-26:
Еще одна версия на проверку: [Candle_iss521_2.zip](https://github.com/Denvi/Candle/files/7609478/Candle_iss521_2.zip)

Вид настроек представлен ниже.
Флаг "Приостанавливать..." на ESP32 пока не проверялся.

[image]

DEST1981, 2021-11-29:
Протестировал.
Смотрите отловил некий баг/фича:
1. Запускаю программу.
2. Ловим М6.
3. Инструмент отправляется в 0 для замены.
4. Программа встаёт на удержание.
5. Если сейчас нажать прервать, а потом удержание. Выдаётся сообщение что программа завершена, но щуп всё равно ищет стол.
В остальном пока хорошо. Прогоню пару плат отпишусь по результатам.

Denvi, 2021-12-05:
Проверил работоспособность на ESP32 с прошивкой Grbl 1.3a, подключение - USB, проблем с зависаниями не обнаружил.
Здесь версия с исправлениями для 4-ех осей: [Candle_iss521_3.zip](https://github.com/Denvi/Candle/files/7656038/Candle_iss521_3.zip)

XRay3D, 2021-12-06:
> Проверил работоспособность на ESP32 с прошивкой Grbl 1.3a, подключение - USB, проблем с зависаниями не обнаружил.
> Здесь версия с исправлениями для 4-ех осей: [Candle_iss521_3.zip](https://github.com/Denvi/Candle/files/7656038/Candle_iss521_3.zip)
> 
> 

А есть ли возможность обновить репозитории, так как я в вашей программе доработанной мной управляю частотником шпинделя через модбас. Появилась необходимость 4ой оси.

XRay3D, 2021-12-07:
Благодарю.

DEST1981, 2021-12-15:
> 

Спасибо всё работает, карта строится по Z.

DEST1981, 2022-01-24:
Добрый день. Прогнал весь цикл. К сожалению поведение программы не адекватное.
1я обнаруженная проблема если сначала запустить z-щуп затем обнулить координаты, затем поднять инструмент с шагом "непрерывно", инструмент непрерывно и поднимается на кнопки не реагирует. Помогает только сброс. С указанием конкретного расстояния работает нормально.
2. В середине работы пришлось прервать программу и запустить снова примерно из середины, пока не попадалась команда смены инструмена было все ок. Но после замены инструмента программа сошла с ума. Сначала Z уехал вверх. Программу сбросил перезапустил с момента смены инструмента, замена прошла успешно, а вот после уже начались причуды, инструмент вышел за пределы рабочей области. Пришлось снова все сбрасывать, но увы нуль уже потерялся. Все придётся начинать заново.
Версия. Candle_iss521_3.zip