Denvi/Candle #525: carving .......on air?

rcpabon, 2021-12-13:
hello, i'm new to cnc and i finished a project on carveco but when i try to carve it on candle it just carves on air, it moves veeeeeery slowly in a very tiny area, can anyone help guide me, sorry for the trouble and i hope to fix this strange problem.

ondras12345, 2021-12-16:
Maybe your units are wrong. Open your gcode file and look for [`G20` or `G21`](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g20-g21).
From what you are describing, I think you are specifying the movements in inches but the machine is set to millimeters (which could be the default).

rcpabon, 2021-12-17:
the g21 line is empty for some reason it just says g21 the code goes like this, this is just a small part obviously.

G17
G21
G0Z1.000
G0X0.000Y0.000S10000M3
G0X32.526Y6.666Z1.000
G1Z-0.300F240.0
G1X32.547Y6.645F420.0
G2X32.526Y6.715I0.458J0.178
G1X32.526Y6.666
G0Z1.000
G0X33.128Y6.049
G1Z-0.300F240.0
G1X33.141Y6.047F420.0
X33.119Y6.059

ondras12345, 2021-12-17:
> the g21 line is empty

That is correct, `G21` is all you need to switch the machine to millimeters.
However, the coordinates is your `G0`, `G1`, etc. movements seem to change very little. That would indicate that they are possibly meant to be in inches. However, if that's the case, they have a huge offset from origin. Also the feed rates seem to be in millimeters per minute and not inches per minute.

One way or another, this issue is most likely not caused by Candle nor Grbl, but rather by your gcode generator.

rcpabon, 2021-12-17:
but when i run it in candle even though it carves on air it follows the path, what exactly would i need to change so that it carves properly? i'm using carveco to design and generate the G-code.

ondras12345, 2021-12-17:
I assume that by it carving on air, you mean that it's moving at the wrong Z level (height). You can fix that easily by pressing "Zero Z" after you jog the machine so that the tool just touches the surface of whatever you are trying to carve. That makes the surface of the material Z=0. Of course, that's assuming that your gcode program expects the surface to be Z=0.

> it follows the path

I thought you said that it was moving in a very tiny area. That would indicate incorrectly set units, probably somewhere in your carveco design. I've never used carveco, so I can't help you with that.

rcpabon, 2021-12-17:
yes, it did at first, i resetted candle and now for some reason it does, seriously i don't know why :( sorry for being a noob at this.

ondras12345, 2021-12-17:
Ok, maybe your gcode is correct then.
Try setting the correct origin of the Z axis with "Zero Z" (procedure described in my previous post) and see if that helps.

rcpabon, 2021-12-20:
I managed to carve by resetting candle and calibrating everything again, I have no idea why it works now but it does, still, thank you for taking your time to help me I appreciate it.

closed 2022-05-07 by Denvi